FreeCAD/FreeCAD #17355: Crash when selecting any axis

sputtys, 2024-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a fresh download (FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage)
I cleared all custom configs by deleting .config/Freecad and /.local/share/Freecad
I created a new body with New->PartDesign->CreateBody
Then I display the origin (toggled with spacebar).
By clicking (or double clicking ) on an axis (z-axis for example) freecad  crashes immedately. 
This was not the case in FreeCAD_1.0.0RC1-conda-Linux-x86_64-py311.AppImage

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-20:
I cannot reproduce on Windows.
Can you try if this issue is still in the latest dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.50
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

xtemp09, 2024-10-20:
No crash here.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Debug
Branch: master
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

sputtys, 2024-10-20:
I tried with FreeCAD_weekly-builds-38991-conda-Linux-x86_64-py311.AppImage
No Crash in this version

```
OS: Ubuntu 24.04.1 LTS (X-Cinnamon/cinnamon/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

maxwxyz, 2024-10-20:
closing this then.

closed 2024-10-20 by maxwxyz

xtemp09, 2024-10-20:
@sputtys, could you upload the backtrace for posterity?

sputtys, 2024-10-20:
yes, where i can find/create it?

xtemp09, 2024-10-20:
@sputtys, when you launch _FreeCAD_ in console, _Linux_ outputs the backtrace into the terminal. One can also launch _FreeCAD_ in _gdb_ or _lldb_.

sputtys, 2024-10-20:
with FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage:
```
sputty@ip5:~/Tools$ ./FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: Datei oder Verzeichnis nicht gefunden
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7f17c9245320]
#1  0x7f1798151fc6 in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /tmp/.mount_FreeCArPHZyE/usr/lib/./libTKBRep.so.7.7+0x26
#2  0x7f1798163b21 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /tmp/.mount_FreeCArPHZyE/usr/lib/./libTKBRep.so.7.7+0xd1
#3  0x7f1798164d35 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /tmp/.mount_FreeCArPHZyE/usr/lib/./libTKBRep.so.7.7+0xd5
#4  0x7f17ac2eddce in Measure::Measurement::length() const from /tmp/.mount_FreeCArPHZyE/usr/lib/Measure.so+0x15e
#5  0x7f17996d0de7 in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCArPHZyE/usr/lib/MeasureGui.so+0x947
#6  0x7f17996d19ee in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCArPHZyE/usr/lib/MeasureGui.so+0x1e
#7  /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f17c98eb6f2]
#8  0x7f17c98ef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5+0x2a
#9  0x7f17c98e064d in QObject::event(QEvent*) from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5+0x7d
#10  0x7f17ca7703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#11  0x7f17cc569079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libFreeCADGui.so+0x99
#12  0x7f17c98b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5+0x108
#13  0x7f17c9909f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5+0x381
#14  /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x7f17c990a75c]
#15  /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f17c7b10ad7]
#16  /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f17c7b13ac7]
#17  /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f17c7b140ce]
#18  0x7f17c990aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5+0x66
#19  0x7f17c98b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5+0x11b
#20  0x7f17c98bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libQt5Core.so.5+0x82
#21  0x7f17cc4c84e1 in Gui::Application::runApplication() from /tmp/.mount_FreeCArPHZyE/usr/bin/../lib/libFreeCADGui.so+0xa41
#22  /tmp/.mount_FreeCArPHZyE/usr/bin/freecad(+0x789f) [0x650dad36089f]
#23  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7f17c922a1ca]
#24  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7f17c922a28b]
#25  /tmp/.mount_FreeCArPHZyE/usr/bin/freecad(+0x7c12) [0x650dad360c12]

```