FreeCAD/FreeCAD #17357: Click on axis in combo view crashes Freecad

hdldesigner, 2024-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a new "Parametric Part" file clicking any of the Body->Origin-> axis, crashes Freecad

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * DynamicData 2.62.0
  * OpenTheme 2024.9.1
  * Glass
  * Gridfinity 0.5.0
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]
[mac_crash_report.log](https://github.com/user-attachments/files/17450312/mac_crash_report.log)
[FreeCAD.log](https://github.com/user-attachments/files/17450315/FreeCAD.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-10-20:
The issue #17355 is related.

luzpaz, 2024-10-20:
@hdldesigner looks like you'll need to use the [weekly](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) or wait for RC3 to come out.

Thanks @xtemp09 for the heads up. Closing as duplicate

closed 2024-10-20 by luzpaz