FreeCAD/FreeCAD #17359: polar pattern of sweep crashes program

Thom821, 2024-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

made a additive pipe (sweep along curve). Trying to create a polar pattern of the pipe immediately crashes the program.

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * sheetmetal.backup1701024896.930162 0.2.62
  * A2plus.backup1692193044.934858 0.4.60k
  * Curves.backup1693340620.716107 0.4.59
  * BIM.backup1692193071.964721 2021.12.0
  * LCInterlocking.backup1701024875.883337
  * fasteners 0.5.29
  * fasteners.backup1698406351.503953 0.4.59
  * Assembly4 0.50.15
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-20:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.
- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

luzpaz, 2024-11-23:
@johnocfii can you reproduce on macos ? TIA

JohnOCFII, 2024-11-23:
I was not able to recreate a crash, but I also did not succeed at creating a polar pattern of the additive pipe.  Without details from the OP, I may not have followed the same steps, as there are many options to take as you create both the additive pipe and the polar pattern.

I've attached a screen shot of both the successful additive pipe, as well as the screen where I attempted to create the polar pattern, and received the message, `15:53:38  PolarPattern: Bnd_Box is void` Also - this was tested with the 1.0 release version, and not the RC2 as the OP used.

```
OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.31
  * Curves 0.6.51
```
[image]

[image]

tarman3, 2024-11-24:
Can not reproduce in ArchLinux
<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39240 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```
</details>

drcowsley, 2024-12-19:
On this theme, I selected polar patter expecting, as in previous versions, to select the feature I wanted repeating from a list.  There was no such list, and because I had always done it that way I wasn't familiar with the method of adding a feature to the empty list.

So I'm not sure what else I did, but I I lost the panel where I put parameters in, tried getting panels back and generally made a mess of any audit track BUT .....

I ended up in a state where any tool I selected when editing a sketch (eg arc) could not be dropped by escape, but it could be replaced by the selection of another tool.  I couldn't get back just the cursor.

That un-usability issue was not cured by closing the sketch or closing the model file.  I had to shut down Freecad and restart it.

So polar pattern called with no feature of the tree as argument breaks the FreeCad program, not the model.

Now that I know about it I'm not asking for help, just trying to help you guys find it.  

Thanks for all you do.

DrC

drcowsley, 2024-12-19:
Another possible clue, when going from 10 occurrences to 11 the rendering on this fails.
[PolarSymmetry 11.zip](https://github.com/user-attachments/files/18203497/PolarSymmetry.11.zip)

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

luzpaz, 2024-12-23:
> Now that I know about it I'm not asking for help, just trying to help you guys find it.

Thanks. Any help you can provide us to reproduce is very appreciated.

drcowsley, 2024-12-26:
It can be embarrassing talking to a current expert on a subject I stopped
following a few decades ago, but back in the day an unset variable was a
serious hazard.  Preventing it was a palliative solution, but working out
how it wreaked its many different forms of havoc was a good way of finding
other weaknesses sooner rather than later.
In this case, whatever was still lurking there, presumably a supposedly set
variable that wasn't, managed to sabotage the structure invoked when I
started thrashing about in the UI resources on panels, which in turn
stopped Escape dropping an edit tool.

How do you stop Escape dropping a line, arc, polygon etc?

Thank you for all you and the team do.  If I see it again I will try to
park what I'm doing and start making notes.

Cheers

DrC

closed 2025-09-27 by maxwxyz