FreeCAD/FreeCAD #17361: Ability to view constraints automatically

Roberto8303, 2024-10-20:
### Original PR

Freecad

### Description and Justification

It would be useful to be able to automatically display once a constraint is selected on the drawing in the constraints table. Now you have to scroll and look for it among everyone and it takes more time.  
[image]
[image]

Roberto8303, 2024-10-20:
they should light up highlighting all the selected ones.
It happens when there are a lot of constraints in a sketch and if you want to hide someone in particular because it disturbs the view of the sketch, pressing the space bar does not hide it and you have to search for it in the constraints list.
so if it would be highlighted automatically you could immediately disable the view with the check mark.

maxwxyz, 2024-10-20:
so the selected constraints is already highlighted in the 3D view and the list. What version are you using, please post full about info.

Roberto8303, 2024-10-20:
they are already highlighted but you have to scroll to find them and see them highlighted. It would be nice if the bar automatically moved to the foreground to show the selected ones.
the version is 1.1dev

Beep6581, 2025-01-03:
> So it should scroll to the selected constraint? What if multiple are selected? related #5877

Yes, it should scroll to the selected constraint item in the QListView. If multiple constraints are selected, scroll to the last-selected one.

PR #18859 (2025-01-21): Auto scroll to constraint on click
When clicking on a constraint in Sketcher, that constraint is now automatically scrolled to in the Constraints listview. If selecting multiple constraints, the last one is scrolled to. Closes #17361
commit a1f5d365 (2025-01-21): Auto scroll to constraint on click (#18859)

* Auto scroll to constraint on click

When clicking on a contrasint in Sketcher, that constraint is now
automatically scrolled to in the Constraints listview. If selecting
multiple constraints, the last one is scrolled to.
Closes #17361

* Check model() for nullptr

* scrollTo only on select, not de-select #18859
closed 2025-01-21 by chennes