FreeCAD/FreeCAD #17365: Internal Involuate Gear will not allow me to increase the number of teeth.  What am I doing wrong?  Read all I could find.

FTCEnergy, 2024-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Internal Involuate Gear will not allow me to increase the number of teeth.  What am I doing wrong?  Read all I could find.

### Full version info

```shell
Can't build planetary gears system without being able to increase the number of teeth.
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Can't build planetary gears system without being able to increase the number of teeth.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FTCEnergy, 2024-10-21:
This what the software shows -- 

18:13:33  Traceback (most recent call last):
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape
    profile = rotate_tooth(tooth, fp.num_teeth)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth
    return part.Wire(flat_shape)
<class 'Part.OCCError'>: BRep_API: command not done
18:13:33  Recompute failed!
18:13:50  Traceback (most recent call last):
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape
    profile = rotate_tooth(tooth, fp.num_teeth)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth
    return part.Wire(flat_shape)
<class 'Part.OCCError'>: BRep_API: command not done
18:13:50  Recompute failed!
18:14:18  Traceback (most recent call last):
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape
    profile = rotate_tooth(tooth, fp.num_teeth)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth
    return part.Wire(flat_shape)
<class 'Part.OCCError'>: BRep_API: command not done
18:14:18  Recompute failed!
18:14:32  Traceback (most recent call last):
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape
    profile = rotate_tooth(tooth, fp.num_teeth)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth
    return part.Wire(flat_shape)
<class 'Part.OCCError'>: BRep_API: command not done
18:14:32  Recompute failed!
18:14:59  Traceback (most recent call last):
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape
    profile = rotate_tooth(tooth, fp.num_teeth)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth
    return part.Wire(flat_shape)
<class 'Part.OCCError'>: BRep_API: command not done
18:14:59  Recompute failed!
18:15:30  Traceback (most recent call last):
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape
    profile = rotate_tooth(tooth, fp.num_teeth)
  File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth
    return part.Wire(flat_shape)
<class 'Part.OCCError'>: BRep_API: command not done
18:15:30  Recompute failed!
18:26:10  PropertyPythonObject::toString(): failed for <class 'freecad.gears.internalinvolutegear.InternalInvoluteGear'>
18:26:10  Traceback (most recent call last):
  File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type InvoluteTooth is not JSON serializable

FTCEnergy, 2024-10-21:
Obviously, there is a bug in the program --

FlachyJoe, 2024-10-21:
Please share the failing file.

FTCEnergy, 2024-10-21:
This what the program shows 

 

18:13:33 Traceback (most recent call last): 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

gear_shape = self.generate_gear_shape(fp) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

profile = rotate_tooth(tooth, fp.num_teeth) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:13:33 Recompute failed! 

18:13:50 Traceback (most recent call last): 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

gear_shape = self.generate_gear_shape(fp) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

profile = rotate_tooth(tooth, fp.num_teeth) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:13:50 Recompute failed! 

18:14:18 Traceback (most recent call last): 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

gear_shape = self.generate_gear_shape(fp) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

profile = rotate_tooth(tooth, fp.num_teeth) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:14:18 Recompute failed! 

18:14:32 Traceback (most recent call last): 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

gear_shape = self.generate_gear_shape(fp) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

profile = rotate_tooth(tooth, fp.num_teeth) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:14:32 Recompute failed! 

18:14:59 Traceback (most recent call last): 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

gear_shape = self.generate_gear_shape(fp) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

profile = rotate_tooth(tooth, fp.num_teeth) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:14:59 Recompute failed! 

18:15:30 Traceback (most recent call last): 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

gear_shape = self.generate_gear_shape(fp) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

profile = rotate_tooth(tooth, fp.num_teeth) 

File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:15:30 Recompute failed! 

18:26:10 PropertyPythonObject::toString(): failed for <class 'freecad.gears.internalinvolutegear.InternalInvoluteGear'> 

18:26:10 Traceback (most recent call last): 

File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps 

return _default_encoder.encode(obj) 

File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode 

chunks = self.iterencode(o, _one_shot=True) 

File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode 

return _iterencode(o, 0) 

File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default 

raise TypeError(f'Object of type {o.__class__.__name__} ' 

<class 'TypeError'>: Object of type InvoluteTooth is not JSON serializable 

18:26:10 PropertyPythonObject::toString(): failed for <class 'freecad.gears.basegear.ViewProviderGear'> 

18:26:10 Traceback (most recent call last): 

File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps 

return _default_encoder.encode(obj) 

File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode 

chunks = self.iterencode(o, _one_shot=True) 

File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode 

return _iterencode(o, 0) 

File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default 

raise TypeError(f'Object of type {o.__class__.__name__} ' 

<class 'TypeError'>: Object of type ViewProvider is not JSON serializable

FlachyJoe, 2024-10-21:
We need the FCStd file to debug, if you can't share it due to copyright then try to recreate the bug in another file.

luzpaz, 2024-10-21:
@FTCEnergy a couple of points of orientation: (please follow these directions)  
* Please respond in github directly by clicking on https://github.com/FreeCAD/FreeCAD/issues/17365 (or else the message formatting breaks and no image or file attachments work correctly
* We need your full [About](https://wiki.freecad.org/About) info
* We need you to send us an example .fcstd file. Zip it up (because github supports .zip files not FreeCAD files) and drag/drop it into github textbox which will auto-attach.

FTCEnergy, 2024-10-21:
I am trying to find the  .fcstd file now. 

 

Call me, if you want. 

Joe Shepard 

(386) 747-3999

FTCEnergy, 2024-10-21:
I sent the requested file as a zip file, but your system refused.  What do you want me to do? 

Joe Shepard 

(386) 747-3999

luzpaz, 2024-10-21:
Not sure why you're not following the steps I wrote out for you? You're still responding via email, See step 1. 
[image]

https://github.com/FreeCAD/FreeCAD/issues/17365#issuecomment-2426456431

FTCEnergy, 2024-10-21:
The error is not with the file. It is with the software. Please take a look at the error the software generates. Does the Gear software have an update that fixes the error? 

Joe Shepard 

(386) 747-3999 

 

 

  

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

 gear_shape = self.generate_gear_shape(fp) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

 profile = rotate_tooth(tooth, fp.num_teeth) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

 return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:13:33 Recompute failed! 

18:13:50 Traceback (most recent call last): 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

 gear_shape = self.generate_gear_shape(fp) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

 profile = rotate_tooth(tooth, fp.num_teeth) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

 return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:13:50 Recompute failed! 

18:14:18 Traceback (most recent call last): 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

 gear_shape = self.generate_gear_shape(fp) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

 profile = rotate_tooth(tooth, fp.num_teeth) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

 return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:14:18 Recompute failed! 

18:14:32 Traceback (most recent call last): 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

 gear_shape = self.generate_gear_shape(fp) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

 profile = rotate_tooth(tooth, fp.num_teeth) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

 return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:14:32 Recompute failed! 

18:14:59 Traceback (most recent call last): 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

 gear_shape = self.generate_gear_shape(fp) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

 profile = rotate_tooth(tooth, fp.num_teeth) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

 return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:14:59 Recompute failed! 

18:15:30 Traceback (most recent call last): 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute 

 gear_shape = self.generate_gear_shape(fp) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\internalinvolutegear.py", line 331, in generate_gear_shape 

 profile = rotate_tooth(tooth, fp.num_teeth) 

 File "C:\Users\Joe\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth 

 return part.Wire(flat_shape) 

<class 'Part.OCCError'>: BRep_API: command not done 

18:15:30 Recompute failed! 

18:26:10 PropertyPythonObject::toString(): failed for <class 'freecad.gears.internalinvolutegear.InternalInvoluteGear'> 

18:26:10 Traceback (most recent call last): 

 File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps 

 return _default_encoder.encode(obj) 

 File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode 

 chunks = self.iterencode(o, _one_shot=True) 

 File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode 

 return _iterencode(o, 0) 

 File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default 

 raise TypeError(f'Object of type {o.__class__.__name__} ' 

<class 'TypeError'>: Object of type InvoluteTooth is not JSON serializable 

18:26:10 PropertyPythonObject::toString(): failed for <class 'freecad.gears.basegear.ViewProviderGear'> 

18:26:10 Traceback (most recent call last): 

 File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps 

 return _default_encoder.encode(obj) 

 File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode 

 chunks = self.iterencode(o, _one_shot=True) 

 File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode 

 return _iterencode(o, 0) 

 File "D:\FreeCAD\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default 

 raise TypeError(f'Object of type {o.__class__.__name__} ' 

<class 'TypeError'>: Object of type ViewProvider is not JSON serializable

FTCEnergy, 2024-10-21:
I responded, and the email was rejected, as I said in a previous email.  A picture does not show the problem.  The problem is with the code.  FreeCAD generates an error.  I sent the whole error that FreeC AD generates.  You are not looking at the error FreeCAD generates.  That's the problem. 

Joe Shepard 

(386) 747-3999 

 

  

 Screenshot_20241021_103251.png (view on web) 

#17365 (comment)

FlachyJoe, 2024-10-21:
@FTCEnergy 

> We need the FCStd file to debug

And follow luzpaz requirements. Any other answer is useless.

FTCEnergy, 2024-10-21:
Your system rejected the original zip file twice.  But, the file is not the 
problem.  Do you not understand that the software is reporting an error no 
matter what.  I created a smaller file that shows the same error.

Attached another file.
Joe Shepard
(386)  747-3999

Roy-043, 2024-10-21:
I can confirm. But it is related the the external freecad.gears workbench.
There is already an issue for this:
https://github.com/looooo/freecad.gears/issues/179

I am therefore closing this one.

closed 2024-10-21 by Roy-043