FreeCAD/FreeCAD #17370: Sketcher carbon copy orientation is not as expected

Rexbas, 2024-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I do a carbon copy, the copy is flipped upside down.

See the video and part.

[box2.zip](https://github.com/user-attachments/files/17451864/box2.zip)

https://github.com/user-attachments/assets/03aebc38-571a-48f5-88d6-3ee1dc7dc54a

### Full version info

```shell
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.0, Qt 5.15.8, Coin 4.0.3, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.23
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Samu152, 2024-10-23:
Can confirm. Seems to happen when the second sketch doen't point in the same direction as the first sketch. If I attach the second sketch to the first sketch directly, the problem goes away.

Rexbas, 2025-08-17:
This is still an issue