FreeCAD/FreeCAD #17371: Freecad 1.0 RC2 Crashes when clicking on a previously generated DatumLine

chof747, 2024-10-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a freecad file which crashes the program whenever I click on a datum line that is placed on the end of my timeline with the following dump:
[dump-datum-line.txt](https://github.com/user-attachments/files/17452074/dump-datum-line.txt)

The datum Line has been created by the intersection of two planes. And whenever i click with my mouse or try to hide it from view, place the cursor on the element above and move down with the cursor ... Freecad crashes. I have attached a small example which reproduces the behaviour. It happens actually every time I try this out even in otherwise complete empty files.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus.backup1726847410.80334 0.4.65 (Disabled)
  * BIM021 2021.12.0
  * Gridfinity 0.6.0
  * A2plus 0.4.67
  * Manipulator 1.5.7
  * kicadStepUpMod 11.3.3
  * fasteners 0.5.29
  * Assembly4 0.50.15
  * Assembly3 0.12.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chof747, 2024-10-20:
[dump-datum-line.FCStd.zip](https://github.com/user-attachments/files/17452085/dump-datum-line.FCStd.zip)
See here a file that is reproducing the error

chof747, 2024-10-20:
One other behaviour I noticed is the following:
1. If I select the line with the mouse in the 3d view first I can change the properties and also click on it in the model treeview, but if I then click again on the 3d view to deselect it, it crashes again.

wwmayer, 2024-10-20:
https://github.com/FreeCAD/FreeCAD/pull/16921

maxwxyz, 2024-10-21:
@chof747 can you check if it still happens in the weekly? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
It should be fixed in the next RC

closed 2024-10-21 by maxwxyz

chof747, 2024-11-15:
Hi @maxwxyz , I tested it with RC4 now and the problem is gone. Thanks a lot.