FreeCAD/FreeCAD #17374: Part, PartDesign: 3D View not updated / broken on empty Compound. Phantom shape displayed and not updated.

AIRCAP, 2024-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In PartDesign it is possible to have an empty/Null shape. For example if there is a body with no Features inside it. This is displayed correctly and Body.Shape.isNull() reports True.

It is however also possible to end up with an empty shape with subtractive features. Easiest way to reproduce.

1. Create a Body
2. Add a Additive Primitive to the Body
3. Add a Subtractive Primitive of same type and shape to the Body

**Expected Outcome:**
The body has no Shape (Body.Shape.isNull()==True) and is not drawn.

**Actual Outcome:** 
1. The body has a shape of type Compound, which is empty
```
Checked object: Body
Shape type:  Compound
Vertices:  0
Edges:  0
Wires:  0
Faces:  0
Shells:  0
Solids:  0
CompSolids:  0
Compounds:  1
Shapes:  1
```
2. An Error is triggered:
`11:47:51  <Part> ViewProviderExt.cpp(1307): Cannot compute Inventor representation for the shape of Unnamed#Cylinder001: Bnd_Box is void`

3. The 3D View shows an outdated "phantom" tessellation of the shape, which may be buggy/broken. This can be triggered for example by temporarily changing the shape of either the additive or the subtractive shape so a remainder remains, and then changing it again such that the shape once more cancels out completely. The previous 3D shape remains displayed as a phantom shape
[image]

It is possible to select or highlight features in this phantom shape in the 3D view, but the geometry (faces, edges) do not actually exist. Exporting the shape creates an empty brelp file or an attampt to make a mesh from it will cause an error (expected) Inspecting the shape in python correctly reveals an empty shape with no vertices, edges, wires or surfaces.

When the file is saved, closed and reopened, the phantom shape is gone. But recomputing the document does not get ri of the phantom shape, only closing and reopening does.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 +58 (Git)
Build type: Debug
Branch: justatest
Hash: 9af05adc1151d7a44a4e55e8d1a31a2744c83bf5
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-21:
Note: Saving the file and reopening it makes  the phantom go away, so I can obviously not provide an example file. The steps to reproduce above must suffice ;)

AIRCAP, 2024-10-21:
Note: Another way to trigger the phantom shape is a "literally" empty compound:

Steps to reproduce:
1. Make a shape
2. Make a compound with the shape as a child
3. Move the shape out of the compound, the compound is now empty

Expected outcome:
Compound 3D view disappears

Actual outcome
Bnd Box Void error message (same as above) and the compound is still rendered with its phantom/old shape / tessellation, even if that shape no longer exists.  On saving/reloading the file that goes away.

FEA-eng, 2026-10-04:
```
>>> obj.Shape.isNull()
False
```

```
Checked object: Body
Shape type:  Compound
Vertices:  0
Edges:  0
Wires:  0
Faces:  0
Shells:  0
Solids:  0
CompSolids:  0
Compounds:  1
Shapes:  1
```

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260916 (Git shallow)
Build date: 2026/09/16 06:32:59
Build type: Release
Branch: (HEAD detached at a4ce44d33)
Hash: a4ce44d33b7e42e16cb8156058a1894237042ab4
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.0, boost 1_84, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```