FreeCAD/FreeCAD #17382: More Meaningful Errors for Operational Errors by End User.

raegun2k, 2024-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Errors from operations are written in a way that is meaningful in troubleshooting the software, but lacks when a user is troubleshooting a build/design/operation.

### Full version info

```shell
Hi,

I would like to toss in an issue I've long struggled with when using FreeCAD. Specifically that errors make more sense if one digs into the Python source, but not if one is just looking at their CAD model.

To give a specific example:

I created a sketch. Two rectangles, one inside the other. I constrained the outside rectangle dimensionally then I attempted to constrain the wall of the internal rectangle dimensionally to the outside wall(one side of the outside rectangle to have it so that the distance between the two walls would always be a set amount of millimetres. It didn't work, because I was constraining two lines, not points. Okay, fair enough. If that's a limitation/design of the software, not big deal, I can adjust easily as the walls were vertically constrained anyways.

But the error reported was this:

10:39:21  pyException: Traceback (most recent call last): File "<string>", line 1, in <module> <class 'ValueError'>: Datum 5.00 mm for the constraint with index 20 is invalid 10:39:21  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(20,App.Units.Quantity('5.000000 mm'))

What am I, as the engineer working on the sketch, supposed to do with that besides know something went wrong. There are multiple ways that error could be reported that would make it more human/readable. However, a very easy one to correct here would be to adjust this bit:

"Datum 5.00 mm for the constraint with index 20 is invalid"

At the most basic, "index 20" means nothing to me. It's not reflected in the UI, and given the constraint's failure to even get generated,it technically doesn't even exist anymore. Even if it did, let's assume it failed due to an upstream change/formula adjustment. The UI won't tell me what constraint index 20 is.

But the software must know what class of constraint it is, right? Even just changing the output of something like "Datum 5.00 mm for the constraint with index 20 is invalid" to "Datum 5.00 mm for the dimension constraint(index 20) is invalid" would be more useful. And honestly, "datum" could possibly nbe changed to just "value" to make it more accessible but that's just a nitpick on my part.

Basically, the errors still seem better written to troubleshoot the programming of the app versus the creation of objects within it. Obviously the former is important. But the latter is in the end what the software will be used for, isn't it?

So perhaps when writing the exceptions, consideration of "if I knew nothing about how this software works at the pythonic level, would this error help me figure out why my operation failed in the software?" could be factored as part of the workflow.
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-08-20:
I agree that such error messages aren't user-friendly. FreeCAD often prints multiple red lines (especially in Sketcher or when creating datums) while it could be just a short warning.

FEA-eng, 2025-10-20:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-04-13:
Recently, there have been some PRs making error messages more specific and user-friendly, but of course, the issue can still be relevant in various parts of FreeCAD (especially when it comes to the Sketcher and TNP, I think).

luzpaz, 2026-09-03:
@FEA-eng is there an ideal way to print out all error messages and see how each may be improved ?

FEA-eng, 2026-09-07:
> [@FEA-eng](https://github.com/FEA-eng) is there an ideal way to print out all error messages and see how each may be improved ?

I don't know, but considering the number and variety of error messages (with some coming straight out of code), it might be necessary to look at the most common cases individually.