FreeCAD/FreeCAD #17387: Sketcher: No tangent autoconstraints for arcs and circles anymore

FEA-eng, 2024-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While testing #15938, I've noticed much worse issue - there are no tangent autoconstraints for arcs and circles anymore.

1.1.0 dev:

[image]

0.21.2:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-21:
@PaddleStroke @AjinkyaDahale @FlachyJoe fyi

AIRCAP, 2024-10-22:
There was a bug in Sketcher where tangent autoconstraints would erroneously be added when connecting a line to a vertex of a bspline - instantly overconsraining it. I don't know who worked on that, but could this issue be a result of trying to fix that bug?

FlachyJoe, 2024-10-22:
It's probably due to #16985. If someone has the time to compile with this commit reverted he could confirm.

closed 2024-10-25 by chennes