FreeCAD/FreeCAD #17392: PartDesign: Applying a pattern to a curved pipe can produce incorrect geometry

Jarno-de-Wit, 2024-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Applying a PartDesign pattern (polar pattern, linear pattern, mirror) to a pipe can sometimes lead to incorrect geometry.

The basic shape of the additive pipe:
[image]
The pipe after applying a linear pattern of 40 mm overall length, 2 occurences, along the BaseX axis:
[image]
The pipe after applying a Mirror pattern around the YZ-plane:
[image]
The pipe after applying a 360 deg total angle, 2 occurences Polar pattern around the base Z-axis:
[image]

Strangely enough, this problem does not occur if a different order is applied, where the bent section is removed from the guide-line sketch, then the pattern is applied, and then the bend section is reintroduced. In this case, instead the following (correct) result is obtained:
[image]

### Full version info

```shell
System 1:
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
  * A2plus 0.4.68

System 2 (used for verification by running Macro's):
OS: macOS Mojave (10.14)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Render 2024.9.25 (Disabled)
  * A2plus 0.4.68
  * Curves 0.6.50
```

### Subproject(s) affected?

PartDesign

### Anything else?

The following zip file contains both the .FCStd file, and the .FCMacro file that creates bugged geometry.
[PipeTest.zip](https://github.com/user-attachments/files/17465734/PipeTest.zip)
The following macro does successfully create the polar pattern:
[Run7.FCMacro.zip](https://github.com/user-attachments/files/17466048/Run7.FCMacro.zip)
Please note: This macro is slightly modified from a recorded macro, since the recorded macro could not be run without errors. The steps taken are the same between the original (recorded) macro, and this version.

The following file contains the .FCStd file which has the bugged geometry, as created in Release Candidate 2.
[PipeTestBugged.zip](https://github.com/user-attachments/files/17465794/PipeTestBugged.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-21:
@maxwxyz - yes, #17119 is quite likely to affect this behaviour. hopefully for the better ;)

Jarno-de-Wit, 2024-10-21:
It seems subtractive pipes are also affected by this bug. The following pictures show the same pipe as above as a subtractive pipe, embedded within a 100 mm x 100 mm x 30 mm box.

The shape before applying a polar pattern:
[image]
The shape after applying a polar pattern:
[image]

The FCStd file used in the screenshots:
[SubtractivePipeTest.zip](https://github.com/user-attachments/files/17466187/SubtractivePipeTest.zip)

The behaviour also occurs on the latest weekly build:
```
  OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
  * A2plus 0.4.68
```

Additionally, it seems the issue can be resolved by manually marking any object placed before the pattern for a recompute, and then recomputing the body. If the `mark for recompute` option is not set for any previous component, the issue persists after a recompute.

AIRCAP, 2024-10-21:
subtractive also works in latest main

AIRCAP, 2024-10-21:
now we need to rule out that its not OCCT version differencet cause all the bad ones were with 7.7.2

AIRCAP, 2024-10-21:
@maxwxyz ok the pipe bug still happens with

```
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Debug
Branch: oldmain
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```
that mans our recent merge spree today fixed it.

AIRCAP, 2024-10-22:
@Jarno-de-Wit I can not reproduce the error with the subtractive pipe. Even with 1.0RC2.

When opening the file, the polar pattern is visibly broken, but that is because the subtractive pipe is also geometrically broken. If you do Analyze Geometry on the subtractive pipe, it reports multiple errors.

If I recompute the subtractive pipe in 1.0RC2 - then these geometry errors are gone, and then also the polar pattern succeeds. This is the case both with and without a polar pattern.

I have not found a way to recreate the situation in the file, was there also a macro involved that switches axis around? If so, please provide that macro

AIRCAP, 2024-10-22:
also can you confirm the error is gone with https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ?

Jarno-de-Wit, 2024-10-22:
While the issue is reduced on the most recent weekly (the previously given test case works correctly), it doesn't seem to be completely gone. Applying the polar pattern around the Base X-axis creates the following geometry:
[image]

Furthermore, mirroring around the XY-plane doesn't seem to do anything at all:
[image]

An axial linear pattern of 41 mm (1 mm longer than the straight section) gives the following result, with a warning that the "Result has multiple solids":
[image]

Here, enabling `AllowCompound` gives two intersecting, yet separate solids:
[image]

System info
```
  OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
  * A2plus 0.4.68
```

All files can be generated from scratch using the following macro (change the number on line 5 to switch between the patterns shown above, or set to 0 (manual) in case you manually want to generate the pattern:
[PipeTest.FCMacro.zip](https://github.com/user-attachments/files/17480719/PipeTest.FCMacro.zip)

Jarno-de-Wit, 2024-10-22:
The issues also occur in case the straight section is left out, and only the arc segment is used.
Arc-only generation macro:
[PipeTestSimplified.FCMacro.zip](https://github.com/user-attachments/files/17480964/PipeTestSimplified.FCMacro.zip)

AIRCAP, 2024-10-22:
Yeah those examples are a bit beyond the capability of OCCT. 

If you have mostly intersecting shapes and attempt boolean operations, there should be a "clear cut" surface where the parts meet. OCCT can then draw new edges on this surface and join the new surface shell.

But if you have different curved shapes, like here pipes that have different curvature, the intersection plane between them is very ill-defined. Even though you start with just two curved surfaces, because they go in and out of each other multiple times, the outcome is chaotic, there's no analytic solution anymore for the resulting shapes and numeric solutions don't converge because they end up with self-intersecting surfaces.

There's a whole section on OCCT limitations here:

https://dev.opencascade.org/doc/occt-7.4.0/overview/html/occt_user_guides__boolean_operations.html#occt_algorithms_10

for example in this fusion:

[image]

the difference between the lower cylinder and the upper pipe becomes infinitely thin at the end - but does not end in the same vertices (which it would if you have the difference exactly the length of the cylindrical part. As such the point where the intersection plane should be is simply ill defined. Any chosen point / edge will leave ill-defined edges around it that have a distance of less than Precision::Confusion() to each other.

The same is true for a polar pattern around x axis, you have two pipes intersecting each other in a way OCCT just can't find a solution.

The only feasible solution would be to discretize the whole surface into triangles, then merge the meshes and turn the result back into solids, but this is something OCCT doesn't do automatically.

While its annoying that OCCT fails in these cases, it can't really be blamed, this is just beyond its capability. Unlike the earlier test cases for there is a well defined and clear analytical solution, here there isn't.

Jarno-de-Wit, 2024-10-24:
Makes sense. In that case, do we want to close this issue, since the most important aspects of this issue have been fixed, and the remaining aspects are in located OCCT?

AIRCAP, 2024-10-26:
I'd say lets leave it open.  There are after all still cases where it fails, and if its closed, other people will just open a new issue with this when they run into it and find no existing open problem report. Instead lets mark it as 3rd party OCCT bug/limitation? @maxwxyz ?

PR #17325 (2024-11-11): Backport PR #17119 into 1.0
fix #17392  (waiting for reporter confirmation)

Backport of #17119 into 1.0.

For now this PR is a draft to test if backporting even works or introduces undesired behavior or compilation errors from a technical side.

Once #17119 is merged in main this can later serve as a PR to consider the backport itself.

Method: 
1. combined all the commits in #17119 up to and including 81e17b76c435453eedae38a2c50e74e03be8f706 into a single commit (diff with main)
2. cherry pick that commit into releases/FreeCAD-1-0

if more commits get pushed into #17119 I would likely force- [... cut]