FreeCAD/FreeCAD #17394: CI: Windows built runs into SketcherApp MaxInt problem

AIRCAP, 2024-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When running CI for #17356 , the Windows build failed with this obscure error while preparing the shape for the actual test:
(Test is for an OCCT issue that occured with very large (kilometer range) parts for which the PR includes workarounds)

No other CI run had that issue. Is there a discrepancy in python version/libraries installed on that build system?

```
2024-10-21T23:33:11.3110304Z OverflowError: Python int too large to convert to C long
2024-10-21T23:33:11.3110834Z 
2024-10-21T23:33:11.3111235Z The above exception was the direct cause of the following exception:
2024-10-21T23:33:11.3111827Z 
2024-10-21T23:33:11.3112024Z Traceback (most recent call last):
2024-10-21T23:33:11.3113004Z   File "C:\FC\build\release\Mod\PartDesign\PartDesignTests\TestHelix.py", line 138, in testGiantHelix
2024-10-21T23:33:11.3114564Z     TestSketcherApp.CreateRectangleSketch(gearSketch, (10*(10**exponent), 0), (1*(10**exponent), 1*(10**exponent)))
2024-10-21T23:33:11.3116305Z   File "C:\FC\build\release\Mod\Sketcher\SketcherTests\TestSketcherSolver.py", line 67, in CreateRectangleSketch
2024-10-21T23:33:11.3118056Z     SketchFeature.addConstraint(Sketcher.Constraint("DistanceX", i + 2, 2, corner[0]))
2024-10-21T23:33:11.3118996Z                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-10-21T23:33:11.3120102Z SystemError: <class 'Sketcher.Constraint'> returned a result with an exception set

```
test code itself is:

```
    def testGiantHelix(self):
        """ Test giant helix """
        for exponent in range(-1,10):
            body = self.Doc.addObject('PartDesign::Body','GearBody')
            gearSketch = self.Doc.addObject('Sketcher::SketchObject', 'GearSketch')
            body.addObject(gearSketch)
            TestSketcherApp.CreateRectangleSketch(gearSketch, (10*(10**exponent), 0), (1*(10**exponent), 1*(10**exponent)))
            xz_plane = body.Origin.OriginFeatures[4]
            gearSketch.AttachmentSupport = xz_plane
            gearSketch.MapMode = 'FlatFace'
            self.Doc.recompute()

            helix = self.Doc.addObject("PartDesign::AdditiveHelix","AdditiveHelix")
            body.addObject(helix)
            helix.Profile = gearSketch
            helix.ReferenceAxis = (gearSketch,"V_Axis")
            helix.Placement = FreeCAD.Placement(FreeCAD.Vector(0,0,0), FreeCAD.Rotation(FreeCAD.Vector(0,0,1),0), FreeCAD.Vector(0,0,0))

            helix.Pitch = 2*(10**exponent)
            helix.Turns = 2
            helix.Height = helix.Turns* helix.Pitch
            helix.Angle = 0
            helix.Mode = 0
            self.Doc.recompute()
 
            self.assertTrue(helix.Shape.isValid())
            bbox = helix.Shape.BoundBox
            self.assertAlmostEqual(bbox.ZMin/((10**exponent)**3),0,places=4)
            # Computed exact value
            # with r = radius, l = length of square, t = turns
            # pi * r**2 * l * t
            expected = pi * ( ((11*(10**exponent))**2) - ((10*(10**exponent))**2) ) * 1*(10**exponent) * helix.Turns
            self.assertAlmostEqual(helix.Shape.Volume/ ((10**exponent)**3),expected/ ((10**exponent)**3),places=2)

```

### Full version info

```shell
???
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Logs can be found in https://github.com/FreeCAD/FreeCAD/actions/runs/11446124641/job/31853110475?pr=17356

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-22:
I can probably work around this by forcing the involved expression to floats. Still weird that only one build system is affected.

FlachyJoe, 2024-10-22:
> Still weird that only one build system is affected.

What I can find is `numpy` related and talk about a 32 bit library on a 64 bit system.

AIRCAP, 2024-10-22:
> > Still weird that only one build system is affected.
> 
> What I can find is `numpy` related and talk about a 32 bit library on a 64 bit system.

I found mentions about that unrelated to numpy too, but it's all a bit vague and unclear. I managed a workaround by forcing the value to float in the test, but this is still odd.