FreeCAD/FreeCAD #17396: Issue when clicking navigation cube too quickly

Samu152, 2024-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://forum.freecad.org/viewtopic.php?p=787568&hilit=navigation+cube#p787568

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Rexbas, 2024-10-22:
That happens because when you click any of the navicube buttons, the active animation is interrupted and a new one is started. The new animation animates from the current orientation to current + 45 degrees. Maybe I can modify the animations to support changing the target orientation while an animation is already started. Then you can quickly click two times and still have a perfect 90 degree rotation.

Samu152, 2024-10-22:
That would be great. I think that would definitely make the cube feel a bit more intuitive.

PR #19719 (2025-02-24): Gui: Accumulate orientation for NaviCube flat button animations
Fixes #17396 

https://github.com/user-attachments/assets/1f437dd4-35c8-4c91-8fbe-077b8339a608