FreeCAD/FreeCAD #17398: Assembly: Failing if assembly placement is modified

PaddleStroke, 2024-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you modify the placement of an assembly, say you rotate it a little bit, then many things go wrong : 
Joints are not finding the correct placement.
Assembly dragger fails to move objects correctly.
Probably more issues.

Perhaps we should hide the Placement property of AssemblyObjects. And also we could prevent the use of transform on the assembly. 
Thoughts?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39026 +5 (Git)
Build type: Release
Branch: core_associate_view
Hash: 50babf4d32fa83a1e3ad099b85ef0207cb883c27
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.60k
  * Assembly4 0.50.2
  * BillOfMaterials-WB-main 0.0.8.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * fasteners 0.5.0
  * FreeCAD - Shortcut.lnk
  * freecad.gears 1.2.0
  * FreeCAD_FastenersWB-master 0.4.56
  * kicadStepUpMod 10.23.0
  * Ondsel 2024.9.13.02
  * OpenDark 2023.12.17
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-22:
So does it fail also when I create an assembly and then use this as a sub assembly in a different or the same file? Or is it just failing when the placement property of the root level assembly of a file is altered?

pierreporte, 2024-10-22:
> Since there're no real usecase to move an assembly.

This is only true for an assembly that happen to be the root of the document.

tomster, 2024-11-09:
I can definitely confirm this report.In particular, once the assembly has been moved (via Transform) the indicators of the joint connectors i.e. when creating a fixed joint seem to be offset by the distance that the assembly has traveled thus making it very difficult and sometimes even impossible to create a working joint.

I've added a minimal assembly that demonstrates this behavior. I.e. try constraining the top of the box with a fixed joint and you can see what I've described. If you reset the placement of the assembly to [0,0,0] everything works again, as expected.

Hope this helps.

[Issue17398.zip](https://github.com/user-attachments/files/17689094/Issue17398.zip)

PaddleStroke, 2025-10-23:
yes

luzpaz, 2025-11-19:
Any further discussion ?