FreeCAD/FreeCAD #17399: Sketcher: "Translate parameters" in array function incorrectly says "Copies" rather than "Columns"

naich58, 2024-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

When "Move / Array transform" is selected on a highlighted item, the "Translate parameters" box has the "Colums" box marked as "Copies".

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.12.2
  * Manipulator 1.4.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-23:
@PaddleStroke is it really columns or is copies correct?

PaddleStroke, 2024-10-23:
Well both are corrects I guess. I'm ok with both. It would make sense to use columns since we use rows under. I used copies because if you just want to make a linear pattern then copies makes sense.

Also if you leave 0 in the copies, it just moves existing geos, which is conceptually sound. If we changed to column, then column = 0 wouldn't make much sense.

Either way is ok for me. Perhaps others can weight in.

maxwxyz, 2024-10-23:
yes, columns makes no sense if you want to just transform the selection. Then 0 copies would be correct.
@FreeCAD/design-working-group any input?

pierreporte, 2024-10-23:
It’s not clear at first sight. Maybe with a radio button to either just translate or make an array (same for rotate and scale). Overall I don’t really like the UX for this. I feel that they are two very different operations and should be separate features.

luzpaz, 2025-08-22:
What's the verdict here ?