FreeCAD/FreeCAD #17403: Exporting to PDF resets hatch rotation

narical, 2024-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When exporting PDF from TechDraw sheet, if there was a section cut and hatch angle was changed manually - it resets to 45 degrees, both in sheet and exported pdf file. Workaround is "print to PDF", that way hatch remains unchanged.

To replicate:
1) Create body in Part Design
2) Create TechDraw sheet
3) Insert a view with body
4) create section from that view
5) change hatch rotation to 45 degrees
6) select sheet, File - Export to PDF

Hatch rotates

### Full version info

```shell
FreeCad v1.1.0dev / 2024/09/20
Windows 10 x64 build 19044
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-22:
Full [About](https://wiki.freecad.org/About) info please. TIA

WandererFan, 2024-10-22:
I'm assuming this is an svg hatch and not a PAT or bitmap??  Which pattern file?

luzpaz, 2024-10-23:
@narical :point_up_2:

semhustej, 2024-10-23:
Hi,

I can confirm the issue.

- it happens with TechDraw_Hatch - svg hatch
- sample file 
[FC_techdraw_hatch_rotation.zip](https://github.com/user-attachments/files/17489473/FC_techdraw_hatch_rotation.zip)

See behavior in the video:

https://github.com/user-attachments/assets/520c0396-eee8-4480-babb-cc00d7846b6e

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

narical, 2024-10-23:
> @narical 👆

The thing is, I discovered this bug in Ondsel dev version, so I tried it inFreecad too. At the moment, I had some (not the latest) dev build. I've decided that most probably version isn't that relevant and this bug could be easily replicated in latest dev Freecad build. Seems I was right. Sorry for late reply, RL+work.

luzpaz, 2024-10-23:
Does anyone know if this is a regression from 0.21 or 0.22 ?

semhustej, 2024-10-23:
> Does anyone know if this is a regression from 0.21 or 0.22 ?

I tested in 0.21.2

- the exported PDF has wrong hatch rotation in same way - **the bug is not a regression**
- inside of FreeCAD, the hatch representation stays the same (I had to redraw the view in FC1.1 to get rid of artefacts). This is a regression but I guess it is related to the first point.

Note: my tests are done with draft elements (wires).

WandererFan, 2024-10-23:
This is a known issue.  On export the svg hatches are converted to bitmaps due to a lack of clipping support in QtSvg.  Qt recently announced that QtSvg will add clipping support in Qt6.  Once that happens, we can get rid of the bitmap conversion.

There is a redraw missing somewhere to set the screen display of the hatch back to the svg rendering.

WandererFan, 2024-10-23:
There is no way that this is getting fixed for 1.0.

closed 2024-10-28 by chennes