FreeCAD/FreeCAD #17404: Problem with face/edge naming, impacting the Assembly

FredLaCasemate, 2024-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Unexpected behavior occurs when joining two parts using edges and then modifying one of the parts. This doesn't happen every time but I'm attaching an example file.
In this example, in the first body there is a sketch intended to make a hole.
When you make this hole, the assembly breaks.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * Assembly4 0.50.13
  * CurvedShapes 1.0.9
  * Curves 0.6.39
  * ExplodedAssembly
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * LCInterlocking
  * sheetmetal 0.4.18
```

### Subproject(s) affected?

Assembly

### Anything else?

[Exemple bug FreeCAD.zip](https://github.com/user-attachments/files/17476721/Exemple.bug.FreeCAD.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-23:
@FredLaCasemate can you retest on v1.0.0 ? pls dont forget to add the full About info, tia!

FredLaCasemate, 2024-11-27:
Hello,
It seems that the bug was fixed in 1.0.0 version. I was not able to reproduce it.
Thanks.
Fred

closed 2024-12-03 by luzpaz