FreeCAD/FreeCAD #17405: Problem with Boolean union to create fusion on recompute

jrb0411, 2024-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After arraying a compound an attempt is made to union the resulting array into a fusion using the Part workbench (Part->Boolean->Union).  The fusion is created but when re-computed the program hangs (not responding).   No errors are produced.  Task manager shows activity.  I have let this run for several minutes with no results.  A zip of the model is included.  I could not find any similar issues.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.48
  * ExplodedAssembly
  * Woodworking 0.21.2.33771
```

### Subproject(s) affected?

None

### Anything else?

[HeatShieldFCv1.zip](https://github.com/user-attachments/files/17477496/HeatShieldFCv1.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-22:
@AIRCAP fixed by your PRs?
@jrb0411 can you please try with the latest weekly build?

jrb0411, 2024-10-23:
I downloaded the latest build this morning and no change.  The program still hangs on the fusion recompute.  I noticed the FreeCad version hasn't changed.

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) 
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.48
  * ExplodedAssembly
  * Woodworking 0.21.2.33771

luzpaz, 2024-10-23:
@jrb0411 you retested RC2 (which doesn't change). The ask was for you to test on the weekly build (bleeding edge development build that gets generated several times a week): https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Can you test and report if you reproduced? Also don't forget to paste full About info of the weekly build you tested. TIA

jrb0411, 2024-10-24:
I downloaded and installed the latest weekly build this morning.  Same result with the attached model.  With an array from the Part workbench  (Part->Boolean->Union).  I noticed the CPU and memory in task manager spiked then settled down. FreeCAD was unresponsive after several minutes.

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.65
  * Assembly4 0.50.15
  * BIM021 2021.12.0
  * ExplodedAssembly
  * woodworking 0.21.2.33771

AIRCAP, 2024-10-25:
With the provided part I have the same issue on latest dev/main if you try to fuse the whole compound that's around 340000 shapes and close to 10000 faces in 40 solids all needed to be merged.

I think the reason it seems to succeed at first is because OCCT doesn't initially compute the shape, only when its explicitly recomputed. So the compound is still shown. If the shape then is computed it just runs into computational complexity.

Since its busy within OCCT code I don't think we can do too much. Probably the solution is to redesign the shape, for example by cutting out an array of stencils out of a single hollow cylinder instead of stacking individual rings.

AIRCAP, 2024-10-25:
Here is an alternate way to construct the part. Instead of assembling pre-cut sections, the holes are cut with an array of stencils out of a single large hollow cylinder. That is still slow but succeeds.

The issue with these parts is that the stencils have overlapping cylindrical sections. OCCT cannot fuse these very well and they result both in Geometry Check errors (dis-continuity warnings) as well as exceeding computational complexity. You should really only fuse or cut multiple curved sections if you have to.

problematic areas marked in red:

[image]

aside from making OCCT crazy it would also make the part really annoying to machine properly

jrb0411, 2024-10-28:
Thank you for the analysis of the issue.  I must confess that I am new to FreeCAD having come from a strong Revit background.  I am following some online tutorials.  Just clicking where they say to click.  I appreciate deeply the committed community to FreeCAD.  I will study your corrections and hopefully elevate my learning of FreeCAD.

closed 2025-09-05 by github-actions[bot]