FreeCAD/FreeCAD #17409: Using PLACEMENT with multiple documents open redirects the current display to the first document

fester225, 2024-10-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When there are multiple documents open and use PLACEMENT, the document displayed changes to the first document opened, not the one you're working on. The solution is to close all documents you are not currently working on. PLACEMENT should not affect which document is displayed.

### Full version info

```shell
[code]
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Rexbas, 2024-10-22:
I have experienced something similar a couple of days ago on 1.0RC2 on Windows 11. I created three new documents, two for parts and one for an assembly containing the two other parts. Whenever I pressed space to toggle visibility, the first document was shown. Also when I double clicked _Assembly_ in the tree view it switched to the first document. The strange thing is that when I open the files now, I cannot reproduce it anymore...

fester225, 2024-10-22:
Thank you. At least it's not only me.

luzpaz, 2024-10-22:
@Rexbas can you share your full [About](https://wiki.freecad.org/About) info that you confirmed this issue on? TIA

fester225, 2024-10-22:
[code]
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Help 1.0.3
[/code]

luzpaz, 2024-10-22:
@fester225 We aren't making changes to 0.21.x. Moving forward we're only addressing RC bugs and 1.1-dev bugs. 

Please start using RC2 if you're going to report bugs, TIA  
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2

Rexbas, 2024-10-23:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL) -> **It is English**
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1