FreeCAD/FreeCAD #17412: 1.0RC2 cannot solve sketch created in stable version without issues

ernstschnell75, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening the attached file created in FreeCAD 0.21.2 in RC1.0.0R2 shows an inability to solve sketch030, which has been created and used for padding without problems before. The sketch looks okay, and it is not immediately obvious, what is wrong with it. Removing constraints does not change the color, or the status, and removing and re-entering lines does not help either.
Console Output:
21:45:27  Transformed: Result has multiple solids. Only keeping the first.
21:45:27  Sketch030: Solving the sketch failed

### Full version info

```shell
Not working in OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

Created in:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[Rugged Box Assembly.zip](https://github.com/user-attachments/files/17484341/Rugged.Box.Assembly.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Samu152, 2024-10-23:
Can confirm. Seems to be an issue with sketch028 in pad017. Since this sketch somehow got malformed, the rest of the model is also broken.

Samu152, 2024-10-23:
Problem may be TNP related. Many of the external geometries in the sketches have moved around and/or the sketches are malformed. Mainly sketch028 and sketch030. Also fillet007 links to the wrong face.

I fixed the model for you, but the issue should still be adressed.
[Rugged Box Assembly.zip](https://github.com/user-attachments/files/17486575/Rugged.Box.Assembly.zip)

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

adrianinsaval, 2024-10-24:
If there are broken references in the sketch it is expected that it won't solve, I doubt that's a bug. The references breaking should be investigated but keep in mind that the toponaming mitigation is not (and can't be) 100% effective
Does the document recompute correctly in linkstage3?
@CalligaroV should this be considered a bug?

CalligaroV, 2024-10-24:
@adrianinsaval, I did some checks and AFAICS I can agree with @Samu152's analysis: the issue looks related to TNP.

This model is pretty complex (for a bug hunt) therefor I fear that it will take me a big amount of time to figure out what's going on, let alone why it's happening and come out with a possible fix.

ATM, considering also that there are other issues in my list waiting from a week ago or more, all I can suggest is to try to fix them manually.
Later, if I find a solution, it will become an automatic task.

I'll be back as soon as I'll have something to share.

Sorry!

CalligaroV, 2024-11-11:
@adrianinsaval, @ernstschnell75, @Samu152 I think I have some good news:

while testing the code I pushed on a branch in my GitHub repo (https://github.com/CalligaroV/FreeCAD/tree/toponaming-ElementMapVersion-code-from-LS3) for #17124 I noticed that the issue in the OP looks fixed by that same code modification.

Unfortunately that modification isn't enough to restore completely [Rugged Box Assembly.zip](https://github.com/user-attachments/files/17484341/Rugged.Box.Assembly.zip): an issue raises with `Fillet007`.

I'm still trying to figure out what's missing (on LS3 other issues occurs but `Fillet007` gets restored correctly).
I'll keep you updated.

For now all I can say is that probably we are half way on fixing this issue completely.