FreeCAD/FreeCAD #17419: VarSet Property - Can't modify tooltip

codeNinja21, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After creating a property in a varset there is no way to add or modify the tooltip once it's been created.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-23:
Hi @pieterhijma can you reproduce ?

Roy-043, 2024-10-23:
Related: #16313.

pieterhijma, 2024-10-24:
@luzpaz, Yes I can reproduce. As @Roy-043 points out and related to this, the VarSet dialog has an issue that tooltips/documentation are not stored. I've just submitted a PR for that.

However, the issue described here is not a VarSet issue per se (although this is where most user will probably notice it). In general it is not possible to change the documentation/tooltip once a property has been created. (At least I am not aware of it).

As a fix for this issue, I would add a context menu item (right click on property in the Property View) like so:

Add Property
Rename property group
Change documentation
Remove property
-----
[ ] Show hidden
etc.

@codeNinja21, would this solution be intuitive for you?

codeNinja21, 2024-10-24:
Yes, a context menu would be intuitive.

Yahoo Mail: Search, Organize, Conquer

luzpaz, 2024-11-01:
> I would add a context menu item (right click on property in the Property View)

@pieterhijma should we rename ticket then ?

pieterhijma, 2024-11-01:
> > I would add a context menu item (right click on property in the Property View)
> 
> @pieterhijma should we rename ticket then ?

Yes, perhaps: Property View: Can't modify tooltip/documentation

Great that there is a topic VarSet. I hope to have time soon to pick up on all varset issues, so from my point of view, you can leave this in the VarSet topic as I will then notice it.