FreeCAD/FreeCAD #17420: Missing part of a surface in Part Design

3ntr0pyc0in, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Missing part of a surface in Part Design.

[Uploading BugN001.zip…](This) is the file.

https://github.com/user-attachments/assets/e4b1a07e-943d-4f09-816b-92b6e87d703b

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * AirPlaneDesign 0.4.1
  * Assembly4 0.50.15
  * CurvedShapes 1.0.12
  * EM 2.1.1
  * ExplodedAssembly
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * OpticsWorkbench 1.0.25
  * Reinforcement
  * Rocket 3.3.0
  * sheetmetal 0.5.3
```

### Subproject(s) affected?

PartDesign

### Anything else?

This behaviour is present also in:
	
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Samu152, 2024-10-25:
The Problem happens with Fillet015. Changing it from 1mm to 0,999mm fixed it. I noticed that FreeCAD has trouble creating fillets that are the same size or larger than the object they are placed on. For example: 

- Use this file: [FilletError.zip](https://github.com/user-attachments/files/17516707/FilletError.zip)
- Change the fillet size from 0,999mm to 1mm
- An error message pops up. (Because the radius of the fillet is now 1mm (2mm diameter) and the pad is also 2mm high.
It is almost the same in your model, but for some reason no error message was shown. Instead the model corrupts.

Exporting your file as .3mf and opening it in another 3D viewer I get the same issue.

I think this happens because FreeCAD gets confused with the edges the fillet created. This has been an annoyance for me as well for several months now and the workaround of making the fillet 0,001mm smaller doen't seem very elegant to me.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

luzpaz, 2024-10-25:
> I think this happens because FreeCAD gets confused with the edges the fillet created. This has been an annoyance for me as well for several months now and the workaround of making the fillet 0,001mm smaller doen't seem very elegant to me.

This sounds like it's OCC related.