FreeCAD/FreeCAD #17422: Quick Measure Giving Incorrect Results for Imported Step File

LHC-Canada, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I’ve been getting some really strange results with the on screen measure feature in the weekly I am using (39024).  I have attached a FCStd file that shows the issue.

Click on the outer edge of the countersunk holes from left to right.  They are all identical at radius 6.6mm, but going from left to right I am getting - 

Radius: 6.6 mm  Correct
Radius: 3.0 mm  ?????
Length: 59 mm   ?????
Radius 2.6 mm   ?????

This might be related to GAZ082 issue he recently posted an issue about.  https://github.com/FreeCAD/FreeCAD/issues/17201

It seems we both have the issue with imported step files.  

I downloaded and ran the NIST checker on the step file but I don’t have Excell installed on my machine and don’t understand most of the software.  However, I did get an image from the software that looked like the chamfer surfaces of the holes are gone when I loaded the file.  I’ve attached a screen shot of that as well.  Also, I have attached the original step file.  This step file originated from Fusion 360 – I was able to open the original Fusion design file and re-save this part as a step file with the same result.
[MeasureTest.zip](https://github.com/user-attachments/files/17492984/MeasureTest.zip)
[MeasureTestStep.zip](https://github.com/user-attachments/files/17492998/MeasureTestStep.zip)

[NIST.zip](https://github.com/user-attachments/files/17493036/NIST.zip)

### Full version info

```shell
My FreeCad Info:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/ProDark/Qt default
Installed mods: 
  * A2plus 0.4.56a
  * Beltrami 1.1.0
  * Curves 0.4.4
  * Defeaturing
  * ExplodedAssembly
  * fasteners 0.3.44
  * fcgear 1.0.0
  * IconThemes
  * kicadStepUpMod 10.13.0
  * Manipulator 1.4.5
  * QuickMeasure 2022.10.28
  * scripts
  * sheetmetal 0.4.24
* VendorParts
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-23:
cc @GAZ082 could you test this ? TIA

maxwxyz, 2024-10-23:
If this is the same issue please add the info to the other issue and we can close this.

Roy-043, 2024-10-23:
Confirmed (with MeasureTest.FCStd and after importing MeasureTestStep.step). But the problem is related to the object.

It also occurs if you do this:
1. Select the 2nd circular edge from the left.
2. Invoke [Std_SendToPythonConsole](https://wiki.freecad.org/Std_SendToPythonConsole).
3. Paste this in the Python console: `sub.Curve.Radius`
4. Result: `3.00000000000001`

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

GAZ082, 2024-10-24:
@luzpaz i confirm this and also @Roy-043 comment. Also tried with safe mode 😎, same result as my issue.

OS: Arch Linux (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1

GAZ082, 2024-10-24:
I think this would be interesting to be tested with libraries different than ours. Sadly cant run in my machine 0.21, Appimage nor packaged  for some weird issues i'm too lazy to debug.

maxwxyz, 2024-10-24:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/17201

closed 2024-10-24 by maxwxyz