FreeCAD/FreeCAD #17425: IFC Export of custom IFCBoolean Property never True

Grosskopf, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've now learned how to use the BIM workbench to generate IFC Files.
It looks awesome, I love it, thank you for making it part of 1.0
I however noticed one IFC Property during my testing that didn't export to IFC as expected:

When adding a property set called "custom" and a Boolean propery called "MyProperty" inside it with Value "True", exporting it to IFC results in the following Line inside IFC:
`#25=IFCPROPERTYSINGLEVALUE('MyProperty',$,IFCBOOLEAN(.F.),$);`

I was able to reproduce it with an empty file except for a Site

my complete IFC Export looks like this:
```
ISO-10303-21;
HEADER;
FILE_DESCRIPTION(('ViewDefinition [CoordinationView]'),'2;1');
FILE_NAME('test_project.ifc','2024-10-23T16:13:59',('',''),(''),'IfcOpenShell 0.7.0','IfcOpenShell 0.7.0','');
FILE_SCHEMA(('IFC4'));
ENDSEC;
DATA;
#1=IFCPERSON($,$,'',$,$,$,$,$);
#2=IFCORGANIZATION($,'',$,$,$);
#3=IFCPERSONANDORGANIZATION(#1,#2,$);
#4=IFCAPPLICATION(#2,'1.0 build 0','FreeCAD','118df2cf_ed21_438e_a41');
#5=IFCOWNERHISTORY(#3,#4,$,.ADDED.,1729700039,#3,#4,1729700039);
#6=IFCDIRECTION((1.,0.,0.));
#7=IFCDIRECTION((0.,0.,1.));
#8=IFCCARTESIANPOINT((0.,0.,0.));
#9=IFCAXIS2PLACEMENT3D(#8,#7,#6);
#10=IFCDIRECTION((0.,1.,0.));
#12=IFCDIMENSIONALEXPONENTS(0,0,0,0,0,0,0);
#13=IFCSIUNIT(*,.LENGTHUNIT.,$,.METRE.);
#14=IFCSIUNIT(*,.AREAUNIT.,$,.SQUARE_METRE.);
#15=IFCSIUNIT(*,.VOLUMEUNIT.,$,.CUBIC_METRE.);
#16=IFCSIUNIT(*,.PLANEANGLEUNIT.,$,.RADIAN.);
#17=IFCMEASUREWITHUNIT(IFCPLANEANGLEMEASURE(0.017453292519943295),#16);
#18=IFCCONVERSIONBASEDUNIT(#12,.PLANEANGLEUNIT.,'DEGREE',#17);
#19=IFCUNITASSIGNMENT((#13,#14,#15,#18));
#20=IFCDIRECTION((6.12323399573677E-17,1.));
#21=IFCGEOMETRICREPRESENTATIONCONTEXT($,'Model',3,1.E-05,#9,#20);
#22=IFCGEOMETRICREPRESENTATIONSUBCONTEXT('Body','Model',*,*,*,*,#21,$,.MODEL_VIEW.,$);
#23=IFCPROJECT('1jQWPjNCD0UhVSmMMj7kcZ',#5,'Unnamed',$,$,$,$,(#21),#19);
#24=IFCSITE('0uDcgZkBX9vhe3V9krnvua',#5,'Site','',$,$,$,'Site',.COMPLEX.,(0,0,0),(0,0,0),0.,$,$);
#25=IFCPROPERTYSINGLEVALUE('MyProperty',$,IFCBOOLEAN(.F.),$);
#26=IFCPROPERTYSET('3$BEOEoMf7KffK0npq_HpH',#5,'custom',$,(#25));
#27=IFCRELDEFINESBYPROPERTIES('0ofj8WAtH3zxVoVof76wd8',#5,$,$,(#24),#26);
#28=IFCRELAGGREGATES('1o88$TqSD0Hg$w6jhFGTt9',#5,'ProjectLink','',#23,(#24));
#29=IFCBUILDING('1$7EnPeE1A5xmAGMWDU1o5',#5,'Default Building','',$,$,$,$,.ELEMENT.,$,$,$);
#30=IFCRELAGGREGATES('1K_T5bx3vEChiDyomHRE7E',#5,'SiteLink','',#24,(#29));
ENDSEC;
END-ISO-10303-21;

```

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-23:
@ semhustej can you reproduce this issue? TIA

semhustej, 2024-10-23:
I can reproduce the issue. I can reproduce it with **a default property set a default property**. All booleans are exported as False=No.

I am attaching FCstd file and also exported IFC file with single wall:
[FC-property-boolean-export.zip](https://github.com/user-attachments/files/17496338/FC-property-boolean-export.zip)

Property setting in FreeCAD:

[image]

Exported properties in IFC viewer:

[image]

luzpaz, 2024-10-23:
cc @yorikvanhavre

PR #17455 (2024-10-28): BIM: Fixed export of boolean IFC properties
Fixes #17425
commit ccf09f59 (2024-10-28): BIM: Fixed export of boolean IFC properties - fixes #17425
commit e3aa8ef9 (2024-10-28): BIM: Fixed export of boolean IFC properties - fixes #17425
closed 2024-10-28 by yorikvanhavre