FreeCAD/FreeCAD #17426: TechDraw

puma0812, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TechDrav Size not   normal in  designer  only

### Full version info

```shell
I encountered an issue where, for example, I designed a 50 mm x 50 mm cube, but when I try to create the drawing for this cube, the dimensions are completely different. The dimensions shown on the drawing are not the same as what I initially designed.
```

### Subproject(s) affected?

TechDraw

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-23:
* We need your full [About](https://wiki.freecad.org/About) info. (Please follow the instructions within the link).
* We need step by step instructions on how to recreate this issue.
* If you can add an example file that would be idea. Make sure to zip the file and then drag/drop it into the textbox to auto-attach it.

luzpaz, 2024-10-23:
@puma0812 you're using 0.21.2 please migrate to RC2 (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2) and report if you can reproduce

FEA-eng, 2024-10-23:
Please also change the title to something more meaningful like "TechDraw: Wrong dimensions in axonometric view".

However, this seems to be a known limitation, not a bug: https://wiki.freecad.org/TechDraw_LengthDimension#Display_3D_measurement

puma0812, 2024-10-23:
Hello

When I design the realistic size, for example, a 150 mm x 150 mm plate, it has 8 holes, 4 of them are 6 mm and 4 of them are 4 mm. The thickness of the plate is 2 mm. The corresponding plates are 65 mm wide and made of 2 mm thick material, extending 100 mm from the material. Between the two plates, there is a column-like support made of 2 mm thick material, which is 65 mm wide and 40 mm high. But I will attach the image I made as well as the project file.

WandererFan, 2024-10-23:
Using projected geometry as dimension references in a non-ortho view will give numbers like this.  There is special tool for creating dims in this case: https://wiki.freecad.org/TechDraw_AxoLengthDimension

To do this without using the AxoLengthDimension, select the reference points in the 3d view, create the dimension, then change the Measure Type from Projected to True.
[image]

[ParallelEdgeTest.FCStd.NOT.zip](https://github.com/user-attachments/files/17499098/ParallelEdgeTest.FCStd.NOT.zip)

puma0812, 2024-10-24:
"Where should I copy the extracted file (file path)?"

> Using projected geometry as dimension references in a non-ortho view will give numbers like this. There is special tool for creating dims in this case: https://wiki.freecad.org/TechDraw_AxoLengthDimension
> 
> To do this without using the AxoLengthDimension, select the reference points in the 3d view, create the dimension, then change the Measure Type from Projected to True. ![design1_2dprojvs3dtrue](https://private-user-images.githubusercontent.com/3587161/379491353-cfd09794-9b76-48e3-a8ee-0fa1d082e316.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3MjgwNzksIm5iZiI6MTcyOTcyNzc3OSwicGF0aCI6Ii8zNTg3MTYxLzM3OTQ5MTM1My1jZmQwOTc5NC05Yjc2LTQ4ZTMtYThlZS0wZmExZDA4MmUzMTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjNUMjM1NjE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2U2ODc2ODM1OGM0NGUyZTJjNzlmYzQwYmIzZTY4OGE5MzQzZDBmYjVkMjFkYTA0NTM3MzMxNWNlMGMzODhmMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.MKKKrxFm8pkBLJsoaKhWS2esT8y8KHhNR12v6frIeuE)
> 
> [ParallelEdgeTest.FCStd.NOT.zip](https://github.com/user-attachments/files/17499098/ParallelEdgeTest.FCStd.NOT.zip)

WandererFan, 2024-11-02:
There isn't a lot different in this file from the one I attached previously, but FWIW, here is another:
[Állvány_withDims.FCStd.not.zip](https://github.com/user-attachments/files/17607191/Allvany_withDims.FCStd.not.zip)

WandererFan, 2024-11-02:
As a broader comment, I've been taught that dimensioning axo views is only done as a kind of last resort when the information is difficult to convey in a conventional (ortho) view.

There is a discussion here: https://www.eng-tips.com/threads/is-dimensioning-detailing-in-isometric-view-acceptable.157995/

closed 2024-11-02 by WandererFan

puma0812, 2024-11-22:
okay, thank you very much, I'll read it through.