FreeCAD/FreeCAD #17427: FEM: Possible Regression: Freeze at 100% CPU when searching mesh for solid

AIRCAP, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When doing FEM with more than one material, it is necessary to assign materials to solids. 

If Materials are assigned to solids, these are being searched in the mesh

on the console this is printed:

```
Check prerequisites...

Get mesh data for constraints, materials and element geometry...
Materials
Constraint: MaterialSolid --> We're going to search in the mesh for the element ID's.
    ReferenceShape ... Type: Solid, Object name: Compound007, Object label: Compound007, Element name: Solid21

```

Then FreeCAD just sits there with 100% CPU load on all cores and nothing happens...

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Debug
Branch: debugTolerance
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.12
  * CfdOF 1.27.10
  * Curves 0.6.48
  * AirPlaneDesign 0.4.1

GMSH:
Version       : 4.14.0-git-efbef2df5
License       : GNU General Public License
Build OS      : Linux64
Build date    : 20241004
Build host    : frg03
Build options : 64Bit ALGLIB[contrib] ANN[contrib] Bamg Blossom Cairo DIntegration Dlopen DomHex Eigen[contrib] Fltk GMP Gmm[contrib] Hxt Jpeg Kbipack LinuxJoystick MathEx[contrib] Mesh Metis[contrib] Mpeg Netgen Nii2mesh ONELAB ONELABMetamodel OpenCASCADE OpenCASCADE-CAF OpenGL OpenMP OptHom Parser Plugins Png Post QuadMeshingTools QuadTri Solver TetGen/BR TinyXML2[contrib] Untangle Voro++[contrib] WinslowUntangler Zlib
FLTK version  : 1.3.4
OCC version   : 7.6.3
Packaged by   : eprice
Web site      : https://gmsh.info
Issue tracker : https://gitlab.onelab.info/gmsh/gmsh/issues
```

### Subproject(s) affected?

FEM

### Anything else?

What other info do you need?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-23:
cc @marioalexis84 @FEA-eng

AIRCAP, 2024-10-23:
[BrokenWing.zip](https://github.com/user-attachments/files/17496200/BrokenWing.zip)

Here is my reproduction file stripped of anything unneeded (like all the parametric bits that generate the shape)
I left a Fusion and a split to allow shape analysis, both the Fusion and the individual parts pass BOP tests - a boolean intersection of everything just recreates the compound.

I had to make the mesh very coarse to fit into githubs 25 MB upload limit, feel free to remesh. I had the issue also with much finer meshes.

If I did anything wrong with the material assignment, tell me.

If the second material is removed, the FEM analysis works. It doesn#t seem to matter which solid is referred to in the second material.

FEA-eng, 2024-10-23:
> I had to make the mesh very coarse to fit into githubs 25 MB upload limit

No need for that, you can just use https://wiki.freecad.org/FEM_MeshClear

This issue looks similar to what was reported here: https://forum.freecad.org/viewtopic.php?t=89890

AIRCAP, 2024-10-23:
[image]
While FreeCAD is frozen, nothing is written in the main GUI report window, since it is not updated. However on the console terminal FreeCAD was started with this is printed:

```
Check prerequisites...

Get mesh data for constraints, materials and element geometry...
Materials
Constraint: MaterialSolid --> We're going to search in the mesh for the element ID's.
Constraint: MaterialSolid001 --> We're going to search in the mesh for the element ID's.
    ReferenceShape ... Type: Solid, Object name: Unnamed1, Object label: Compound, Element name: Solid3

```

I started it in gdb and did a backtrace - it printed this:

```
#0  0x00007fffc702f1ce in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.cxx:105
#1  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#2  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#3  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#4  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#5  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#6  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#7  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#8  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#9  0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#10 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#11 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#12 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#13 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#14 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#15 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#16 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#17 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#18 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#19 0x00007fffc702f212 in TopLoc_Location::Multiplied(TopLoc_Location const&) const (this=0x7fffffff7e80, Other=...) at ./src/TopLoc/TopLoc_Location.lxx:68
#20 0x00007fffc702f43f in TopLoc_Location::Predivided(TopLoc_Location const&) const (this=this@entry=0x7fffffff7f58, Other=...) at ./src/TopLoc/TopLoc_Location.cxx:146
#21 0x00007fffcc6d2705 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, opencascade::handle<Geom_Surface> const&, TopLoc_Location const&, double&, double&, bool*) (E=..., S=..., L=..., First=@0x7fffffff7fc0: 6.9533434555614543e-310, Last=@0x7fffffff7fc8: 5.9287877500949585e-322, theIsStored=0x0) at ./src/TopoDS/TopoDS_Shape.hxx:91
#22 0x00007fffcc6d2902 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, TopoDS_Face const&, double&, double&, bool*) (E=..., F=..., First=@0x7fffffff7fc0: 6.9533434555614543e-310, Last=@0x7fffffff7fc8: 5.9287877500949585e-322, theIsStored=theIsStored@entry=0x0) at ./src/BRep/BRep_Tool.cxx:304
#23 0x00007fffcc6e11fb in BRepAdaptor_Curve2d::Initialize(TopoDS_Edge const&, TopoDS_Face const&) (this=0x555565f48820, E=..., F=...) at ./src/BRepAdaptor/BRepAdaptor_Curve2d.cxx:79
#24 0x00007fffcc6e12f0 in BRepAdaptor_Curve2d::BRepAdaptor_Curve2d(TopoDS_Edge const&, TopoDS_Face const&) (this=0x555565f48820, E=..., F=...) at ./src/BRepAdaptor/BRepAdaptor_Curve2d.cxx:43
#25 0x00007fffcc9a68bd in BRepTopAdaptor_TopolTool::Initialize(opencascade::handle<Adaptor3d_Surface> const&) (this=0x555568df2680, S=...) at ./src/Standard/Standard_Transient.hxx:35
#26 0x00007fffcc9a6dc5 in BRepTopAdaptor_TopolTool::BRepTopAdaptor_TopolTool(opencascade::handle<Adaptor3d_Surface> const&) (this=0x555568df2680, S=...) at ./src/BRepTopAdaptor/BRepTopAdaptor_TopolTool.cxx:68
#27 0x00007fffcc873872 in IntCurvesFace_Intersector::IntCurvesFace_Intersector(TopoDS_Face const&, double, bool, bool) () at ./src/Standard/Standard_Transient.hxx:35
[... cut ...]
#94 0x00007ffff1b5410b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#95 0x00007ffff1b2e35b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#96 0x00007fffeb2f332e in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#97 0x00007fffef41417d in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0

```

AIRCAP, 2024-10-23:
In 0.21.2 I do not see this behaviour. However there are already differences in mesh generation. In 0.21.2 I see warnings when generating the mesh (and it doesn't load the mesh already generated by 1.1) 

When writing the inp file it then runs into a different error, but it definitely passes the situation that freezes in 1.0:

on console:

```
Get mesh data for constraints, materials and element geometry...
Materials
Constraint: MaterialSolid --> We have mesh groups. We will search for appropriate group data.
  found mesh group for the IDs: MaterialSolid_Volumes, Type: Volume
Constraint: MaterialSolid001 --> We have mesh groups. We will search for appropriate group data.
  found mesh group for the IDs: MaterialSolid001_Volumes, Type: Volume
Count finite elements as sum of constraints:   157240
Count finite elements of the finite element mesh: 157240
True
ConstraintFixed:
    Type: Fem::ConstraintFixed, Name: ConstraintFixed
  found mesh group for the IDs: ConstraintFixed_Nodes, Type: Node
ConstraintForce:
    Type: Fem::ConstraintForce, Name: ConstraintForce
  Warning --> Force = 0
    ReferenceShape ... Type: Face, Object name: Unnamed1, Object label: Compound, Element name: Face13
Error on: ConstraintForce --> Unnamed1.Face13
  sum_node_areas: 104224.56293748526
  ref_face_area:  135533.48756206635
Traceback (most recent call last):
  File "/usr/share/freecad/Mod/Fem/femtaskpanels/task_solver_ccxtools.py", line 312, in write_input_file_handler
    self.fea.write_inp_file()
  File "/usr/share/freecad/Mod/Fem/femtools/ccxtools.py", line 388, in write_inp_file
    meshdatagetter.get_mesh_sets()
  File "/usr/share/freecad/Mod/Fem/femmesh/meshsetsgetter.py", line 152, in get_mesh_sets
    self.get_constraints_force_nodeloads()
  File "/usr/share/freecad/Mod/Fem/femmesh/meshsetsgetter.py", line 330, in get_constraints_force_nodeloads
    femobj["NodeLoadTable"] = meshtools.get_force_obj_face_nodeload_table(
  File "/usr/share/freecad/Mod/Fem/femmesh/meshtools.py", line 1181, in get_force_obj_face_nodeload_table
    ratio = sum_node_load / frc_obj.Force
ZeroDivisionError: float division by zero
```

I assume this is due to the previously triggered meshing issues.

```

OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)

```

FEA-eng, 2024-10-23:
What's interesting, there's no such indefinite slowdown in a simple test compound consisting of two cubes even when the mesh has more nodes and elements than the OP's wing (which has 196 885 nodes and 92 333 elements while my test model has 211 305 nodes and 147 054 elements). It takes some time but doesn't freeze so badly. In such cases, I would suspect some underlying issues with the geometry. Especially when booleans and such are involved. The [older report](https://forum.freecad.org/viewtopic.php?t=89890) is similar in that matter.

[Compound test.FCStd.zip](https://github.com/user-attachments/files/17496640/Compound.test.FCStd.zip)

FEA-eng, 2024-10-23:
You can run a frequency analysis to see what's disconnected. Perhaps a tie constraint will be needed.

But regarding the issue, perhaps it's caused by the TNP code. I suspected that when replying to the forum post linked above.

AIRCAP, 2024-10-23:
There is definitely some difference in how this is done now. The command line output in 0.21.2 is

Check prerequisites...

```
Get mesh data for constraints, materials and element geometry...
Materials
Constraint: MaterialSolid --> We have mesh groups. We will search for appropriate group data.
  found mesh group for the IDs: MaterialSolid_Volumes, Type: Volume
Constraint: MaterialSolid001 --> We have mesh groups. We will search for appropriate group data.
  found mesh group for the IDs: MaterialSolid001_Volumes, Type: Volume
Count finite elements as sum of constraints:   463400
Count finite elements of the finite element mesh: 463400
True
```

In 1.0:

```
Check prerequisites...

Get mesh data for constraints, materials and element geometry...
Materials
Constraint: MaterialSolid --> We're going to search in the mesh for the element ID's.
    ReferenceShape ... Type: Solid, Object name: Compound007, Object label: Compound007, Element name: Solid21
```

so
We have mesh groups. We will search for appropriate group data.
vs
We're going to search in the mesh for the element ID's.

AIRCAP, 2024-10-23:
The code that prints the old message is still there. I think its a change in the gmsh creation feature which seems tohave been rewritten. It no longer creates meshgroups, as such another code branch is triggered. Its possible that the "other" code branch always had this issue. But why are meshgroups not used even if meshgroups are enabled in FEM Preferences?

in 0.21.2 Meshgroups, although marked experimental were greyed out and always enabled - now they are no longer greyed out, but enabling them seems to not actually use them

1.1:
[image]

in 0.21.2 i have that listed as enabled but greyed out

AIRCAP, 2024-10-23:
Whn Mesh groups are disabled by config override, 0.21.2 also freezes. So the bug is old, but 0.21.2 didn#t trigger it since it defaulted to mesh groups.

The **regression** is that mesh groups are now no longer selectable (respectively selecting the feature has no effect)

FEA-eng, 2024-10-23:
Then this PR is related: #12437.

@marioalexis84

AIRCAP, 2024-10-23:
The issue is definitely caused by Mesh creation. If the Mesh is generated in 0.21.2 with mesh groups enabled. I get the following output:

```
  ElementDimension: 3
  Group meshing for analysis is set to true in FEM General Preferences. Are you really sure about this? You could run into trouble!

```

If the file is than saved, opened in 1.1dev  and .inp file written - without remeshing (using the mesh stored in the file) then I get the same output:
```

Get mesh data for constraints, materials and element geometry...
Materials
Constraint: MaterialSolid --> We have mesh groups. We will search for appropriate group data.
  found mesh group for the IDs: MaterialSolid_Volumes, Type: Volume
Constraint: MaterialSolid001 --> We have mesh groups. We will search for appropriate group data.
  found mesh group for the IDs: MaterialSolid001_Volumes, Type: Volume
Count finite elements as sum of constraints:   210306
Count finite elements of the finite element mesh: 210306
True
```

and the bug is not triggered, FEM works. However if I recreate the mesh in 1.1 then meshgroups are not used, regardless whether they are enabled in preferences or not.

AIRCAP, 2024-10-23:
This regression happened prior to 1.0RC1 - 1.0RC1 already ignores the group meshing preference on mesh creation

AIRCAP, 2024-10-23:
im doing a bisect

FEA-eng, 2024-10-23:
> im doing a bisect

Isn’t it just #12437?

AIRCAP, 2024-10-23:
as I understand that PR is just enabling the checkbox and changing the default value. the problem is that the value of the checkbox is ignored by the actual mesh generation code - that must have been changed elsewhere/elsewhen

AIRCAP, 2024-10-23:
one issue with bisect is - 0.21.2 branched off from main in 0.21.0 rc stage - and 0.21.0 does NOT function (it doesn't produce a mesh with mesh groups for some reason) 0.21.2 does -- so i need to first find the commit in the 0.21.2 release branch that enabled this feature, then figure out when it was removed (or maybe never introduced) in main

AIRCAP, 2024-10-23:
OK because I had not activated the FEM Analysis even my previous assessment was wrong. The regression happened AFTER RC1 RC1 is good

marioalexis84, 2024-10-23:
I'm going to prepare the code to fix the regression in meshing groups. I unintentionally disabled it a few pull requests ago.

On the other hand, the freeze is due to the fact that the code used to find nodes in a solid is not fast enough when working on this mesh.

AIRCAP, 2024-10-24:
Yeah, confirmed. By now the bisect is also complete and homed in on cabfbb749fdf5ad822c690a095e62388d9395a2d but you knew that already ;)

AIRCAP, 2024-10-24:
@luzpaz I think you can mark this as confirmed, as there's even a forum thread about it. It's actually two bugs for the price of one: 
1. Searching for solid ID's in certain meshes takes forever - at 100% CPU load ... I don't know if it takes literally forever, but I have waited over an hour once because I forgot about it and it still hadn't progressed - this only happens when meshgroups are disabled. This bug is old, it was already there in 0.21.2
2. The regression: a change in the commit above made it so meshgroups were not used even when they were enabled - triggering bug 1. If I understand correctly, @marioalexis84 is working on a fix already which should probably be tasked for 1.0 considering the regression happened very recently - post 1.0RC1  .  Since this is also a crash - should this be a blocker?

FEA-eng, 2024-10-25:
> After fixing the mesh groups, it manages to write the .inp file (on my machine it takes about 1 minute)

Do you plan to submit a PR for that soon or should this issue be marked as a blocker just in case (only 3 blockers are left now so we are getting very close to 1.0)?

Non-positive jacobians are common and usually solved by switching to Second Order Linear.

AIRCAP, 2024-10-25:
> @AIRCAP are you using the exact same mesh parameters from Gmsh in version 0.21? I see you get 463400 elements, but in the model you share the mesh has about 92000 elements. After fixing the mesh groups, it manages to write the .inp file (on my machine it takes about 1 minute) but the simulation fails due to non-positive jacobian determinant on some elements.

I did not change any parameters when i compared between FC versions. but before I did that I tried everything (including different meshing parameters, to see if that fixes anything - it didn't - the discrepancy is from that)

However when I create the file with 1.0 and then open in 0.21 the mesh object was completely gone - and had to be recreated.
The other way around works, 1.0 does display the mesh object created by 0.21

As I said when I run the mesher in 0.21 and then reopen the file in 1.0 i can write the .inp file.

if i run the mesher in 1.0 without meshgroups i get the freeze.

I already tried (by reverting the cabfbb7 commit in main) if i create a mesh with mesh groups in 1.0 i can run the analysis successfully. (the non-positive jacobi disappears when i explicitly set max and min mesh sizes to have it fine enough)

AIRCAP, 2024-10-25:
@marioalexis84 ah yes for the "shared file" i deliberately reduced mesh size to reduce file size, it wouldn't fit in the github 25 MB limit otherwise.  this change in mesh size didn't fix anything for me, the inp file writing still froze.

AIRCAP, 2024-10-30:
I've been looking into the 100% CPU load side of things a bit more (Without Mesh Regions)

Here is a much simpler file

[smaller.zip](https://github.com/user-attachments/files/17579822/smaller.zip)

it now can write the inp file:

> Get mesh data for constraints, materials and element geometry...
> Materials
> Constraint: MaterialSolid --> We're going to search in the mesh for the element ID's.
> Constraint: MaterialSolid001 --> We're going to search in the mesh for the element ID's.
>     ReferenceShape ... Type: Solid, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Solid1
> **<here is where it takes forever>**    ReferenceShape ... Type: Solid, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Solid2
> **<here is where it takes forever again>**   binary search: get_femelements_by_femnodes_bin
> len femnodes_ele_table: 19167
> found Volumes: 4096
> Count finite elements as sum of constraints:   9922
> Count finite elements of the finite element mesh: 9922
> ConstraintFixed:
>     Type: Fem::ConstraintFixed, Name: ConstraintFixed
>     ReferenceShape ... Type: Face, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Face161
>     ReferenceShape ... Type: Face, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Face88
> ConstraintForce:
>     Type: Fem::ConstraintForce, Name: ConstraintForce
>     ReferenceShape ... Type: Face, Object name: BooleanFragments, Object label: BooleanFragments, Element name: Face490
> Error on: ConstraintForce --> BooleanFragments.Face490
>   sum_node_areas: 1538.6418339695256
>   ref_face_area:  1557.913072171677
> Deviation sum_node_load to frc_obj.Force is more than 1% :  0.9876300940364476
>   sum_ref_face_node_area: 1538.6418339695256
>   sum_ref_face_area:      1557.913072171677
>   sum_node_load:          98.76300940364476
>   frc_obj.Force:          100.0
>   the reason could be simply a circle area --> see method get_ref_face_node_areas
>   the reason could also be a problem in retrieving the ref_face_node_area
> **Getting mesh data time: 10551.376 seconds.**
> 
> CalculiX solver input writing...
> Input file:/tmp/fcfem_w86r2eev/FEMMeshGmsh.inp
> One monster input file.
> Writing time CalculiX input file: 0.185 seconds.
> 

but it is still excessively long, you probably don't want to wait that long ;)

commit 8865450a (2024-10-31): Fem: fix Gmsh analysis mesh groups - fixes #17427
commit 9cfec546 (2024-10-31): Fem: fix Gmsh analysis mesh groups - fixes #17427
closed 2024-10-31 by chennes