FreeCAD/FreeCAD #17428: Tangent constraint to an arc connects to invisible circle of the arc

SmooreInsight, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you make a semicircle arc, for example in a "u" shape and have another sketch object closer to the missing part of the circle that the arc would form if it continued it's path to make a circle, the sketch object will connect to the invisible part of that arc's hypothetical completed circle. For example, if a horizontal line was above a "U" shaped arc, selecting both and hitting the 't' hotkey will connect the horizontal line above the 'U" arc and not along the actual path of the arc.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.61
  * btl 0.9.9
  * fasteners 0.5.0
  * FeedsAndSpeeds 0.5.0
  * Help 1.0.3

To recreate:
Open a new doc.
Create a body
Open a sketch on the XY plane
Make sure that your document and sketch are named "Unnamed" and "Sketch" respectively, or make the appropriate changes to the following code. Otherwise run the following code:

`App.getDocument('Unnamed').getObject('Sketch').addGeometry(Part.ArcOfCircle(Part.Circle(App.Vector(0.000000, 0.000000, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 16.467812), 3.141593, 6.283185),False)
App.getDocument('Unnamed').getObject('Sketch').addGeometry(Part.LineSegment(App.Vector(-20.167097, 17.064562, 0.000000),App.Vector(22.434345, 16.945227, 0.000000)),False)
App.getDocument('Unnamed').getObject('Sketch').addConstraint(Sketcher.Constraint('Tangent',1,0))`

Note that the lint is not tangent to the arc, but rather, tangent to the invisible circle that the arc would complete if it continued along the same curved path.
```

### Subproject(s) affected?

None

### Anything else?

Here is the forum post where I asked for help with this problem as it applies to my current needs:
https://forum.freecad.org/viewtopic.php?t=91582
[Unnamed.zip](https://github.com/user-attachments/files/17496665/Unnamed.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-23:
Not a bug but how it is intended.
https://wiki.freecad.org/Sketcher_ConstrainTangent
https://wiki.freecad.org/Sketcher_ArcOverlay

Closing this.

closed 2024-10-23 by Roy-043

SmooreInsight, 2024-10-23:
@Roy-043 Thanks for the clarification, but can I get some further idea of what's going on? That can't be determined as an optimized way for it to operate, right? Is there an easy way to flip it when it's on the wrong side?