FreeCAD/FreeCAD #17429: draft offset: impossible to offset a BSpline using the keyboard (to enter an exact offset). using the mouse works

kervel, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps:

1. draft WB: draw a BSpline and select it
2. click the draft offset tool
3. move mouse to the correct side
4. in the panel, add offset distance
5. hit enter

get this:

```
21:01:54  Traceback (most recent call last):
  File "/tmp/.mount_FreeCALDaKdK/usr/Mod/Draft/draftutils/todo.py", line 163, in doTasks
    Gui.doCommand(string)
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCALDaKdK/usr/Mod/Draft/draftfunctions/offset.py", line 139, in offset
    newobj.Shape = DraftGeomUtils.offsetWire(obj.Shape,delta,occ=True)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCALDaKdK/usr/Mod/Draft/draftgeoutils/offsets.py", line 250, in offsetWire
    length = abs(dvec.Length)
                 ^^^^^^^^^^^
AttributeError: 'list' object has no attribute 'Length'

21:01:54  ToDo.doTasks, Unexpected error:
<class 'AttributeError'>
in ['offst = Draft.offset(FreeCAD.ActiveDocument.BSpline, [FreeCAD.Vector(-68.26622065253203, -27.529146433486023, 14.698608703474715), FreeCAD.Vector(-49.54848726254038, -10.952756402361173, 21.834685099984917), FreeCAD.Vector(-18.820083885890288, -1.3062011174464614, 11.823370826337682), FreeCAD.Vector(-7.988057326066063, -6.849800104067812, -2.767869693253764), FreeCAD.Vector(11.019345435970411, -2.730617804518765, -11.245868901233019), FreeCAD.Vector(25.56432326018741, 11.49492477538402, -4.037701819020372), FreeCAD.Vector(29.012864541984086, 24.871566300801792, 10.044189253736382), FreeCAD.Vector(45.351727876861666, 35.0586298769704, 10.976551162759907)], copy=True, occ=True)']
```

doing the exact same thing, but now with the mouse and the offset succeeds.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Woodworking 0.21.30486
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-23:
@Roy-043 FYI and maybe related to https://github.com/FreeCAD/FreeCAD/issues/10435 (which could be split up)

Roy-043, 2024-11-29:
This issue is mentioned in #10435 (nr. 6). Closing this as a duplicate.

closed 2024-11-29 by Roy-043