FreeCAD/FreeCAD #17431: Add Navigation Style to About info

luzpaz, 2024-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Since navigation options are getting a lot more support and numerous, it may be of benefit to add a field in the About info regarding the Navigation style that the user has setup.

Example:  
`Navigation: Touchpad` 

Usecase: 
* https://github.com/FreeCAD/FreeCAD/pull/17341#issuecomment-2433209263

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-23:
CC @Rexbas

maxwxyz, 2024-10-23:
@benj5378 FYI

luzpaz, 2024-10-23:
@Rexbas can you provide insight how to pull these variables out via the API ? :stuck_out_tongue:

benj5378, 2024-10-23:
Not sure if this info in about is relevant. If people are having trouble due to the navigation style, then I think they are likely to already be aware it's due to the navigation style (and mention it in the issue). I suggest awaiting if there will be any issues connected to navigation style, where the bug reporter isn't themselves aware and mention it.

luzpaz, 2024-10-23:
Agreed that it's not relevant most of the time but nevertheless it is when it is. Maybe at some point we can split the output into an optional 'basic' and 'verbose' setting. Navigation styles would fall into 'verbose'. But for now i don't see the harm in adding it to the general About info.

luzpaz, 2024-10-25:
Another usecase for this feature:
> Maybe there is a relation with the mouse navigation style (I use CAD)? Can you both please retest with a 'fresh' CFG? 
* Source: https://github.com/FreeCAD/FreeCAD/issues/17461#issuecomment-2437944935

luzpaz, 2024-10-27:
Another usecase:  
[image]
* Source: https://github.com/FreeCAD/FreeCAD/issues/17440#issuecomment-2439664657

luzpaz, 2024-10-27:
Another usecase: 
[image]
* Source: https://github.com/FreeCAD/FreeCAD/issues/6106#issuecomment-2423342525

Rexbas, 2024-10-27:
Another usecase:
> Which mouse navigation do you use?

* Source: https://github.com/FreeCAD/FreeCAD/issues/16549#issuecomment-2352200578

Rexbas, 2024-10-27:
> can you provide insight how to pull these variables out via the API ? 😛

I think the easiest is to use the parameters, like here:

https://github.com/FreeCAD/FreeCAD/blob/dbc91b393c2c96c6ed9302841f73919b4c5986ae/src/Gui/PreferencePages/DlgSettingsNavigation.cpp#L147-L158

PR #21273 (2025-05-19): Add Navigation Style/Orbit Style/Rotation Mode to About info
Closes #17431
commit 65fe3b85 (2025-05-19): Add Navigation Style/Orbit Style/Rotation Mode to About info

Closes #17431
closed 2025-05-19 by hyarion