FreeCAD/FreeCAD #17439: Insert length window cannot be moved

SB-R2D2, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I try to move the "Insert Length" window just to see something behind it, it moves the whole window of the application (and not only the insert length window). Also, the maximize window button disappears.
The maximize windows buttons re-appears after entering a value and clicking OK.
This happens every time.
Is this a bug?
Attached picture.

I use Ubuntu 22.04.5
Freecad 1.0RC2
[image]

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Israel (en_IL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-24:
Is this the same behavior as in the stable 0.21.2? Is it also happening in the latest weekly?

SB-R2D2, 2024-10-24:
> Is this the same behavior as in the stable 0.21.2? Is it also happening in the latest weekly?

I don't know. I am not a developer and I installed only version 1.0. I just put here what occurred to me to improve the software.

luzpaz, 2024-10-24:
I can't reproduce on 
```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

@SB-R2D2 can you install the weekly build and see if you can reproduce ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

RWL-69, 2024-10-24:
Just tested on a quick test and I didn't have any issues with display of any dimension input UI.

Looking at the Hash and release, they are the same weekly release as mine, which is the most current as of this posting.  See below.  I am on Fedora and they are Manjaro so may be related to differences.

I am using Xorg, they may be using Wayland which is known to have issues with KDE. 
 (https://community.kde.org/Plasma/Wayland_Known_Significant_Issues)

OS: Fedora Linux 39 (KDE Plasma) (KDE/plasmax11/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

SB-R2D2, 2024-10-25:
I installed also version 0.21.2 (together with 1.0RC2) and checked on both. I still have this issue.
To create it:
1/ Open new file
2/ Create a new sketch
3/ Add a line (any)
4/ Try to give a length
5/ When the "Insert Length" window shows up, try to drag the window.
6/ All app window drags/moves (and not the small "insert window" alone)

luzpaz, 2024-10-25:
@SB-R2D2 as requested above, can you also kindly test weekly version ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

benj5378, 2024-10-27:
This should be very easy to fix

godiard, 2025-06-01:
Looks like this is related to a Gnome preference https://forum.qt.io/topic/120852/make-modal-dialogs-moveable-again

You can try doing:
```
gsettings set org.gnome.mutter attach-modal-dialogs false
``` 

Probably this is not a FreeCAD bug

luzpaz, 2025-06-01:
Thanks @godiard  
@SB-R2D2 can you confirm ?

closed 2025-06-01 by luzpaz

benj5378, 2025-06-11:
While it can be changed through gnome settings, I believe it is still FreeCAD setting the non-movable state in the first place. The question is more, if it should be changed.

I recall some PRs doing similar windows movable, and other PRs making them unmovable.