FreeCAD/FreeCAD #17440: Linux FreeCAD 1.0 RC2 BIM crashes on medium complex object

hds4, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FC crashes in BIM when adding features (windows in this case) to a house. Attached a strace of the FC session when crashing. Also attached the house model. It works fine in FC 21.2
To reproduce the carsh you may need to place one or more objects onto the house. I could not pin down the crash to a specific action. Sometimes it crashes immediatly (like this case) other times it takes three or four window objects to crash FC.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[HausTest.FCStd.zip](https://github.com/user-attachments/files/17501624/HausTest.FCStd.zip)

strace -qqq ./FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImag

SNIP startup Text.
.......
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Ignore recovery file  "/home/schroete/.cache/FreeCAD/Cache/FreeCAD_Doc_6082cce5-ab80-4458-8ee5-5c99a6dab8a7_cf365a_4062/fc_recovery_file.fcstd"  because it is older than the project file "/home/schroete/HausTest.FCStd" 

Wähle eine Fläche auf einem existierenden Objekt oder wähle eine Voreinstellung

### Selecting the Window here. Placing it on the house crashes FC immediatly ####

Program received signal SIGSEGV, Segmentation fault.
[{WIFSIGNALED(s) && WTERMSIG(s) == SIGSEGV && WCOREDUMP(s)}], 0, NULL) = 4509
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f69bf220050}, {sa_handler=0x55ddf0d0b0b0, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f69bf220050}, 8) = 0
ioctl(2, TIOCGWINSZ, {ws_row=40, ws_col=151, ws_xpixel=1514, ws_ypixel=925}) = 0
openat(AT_FDCWD, "/usr/share/locale/locale.alias", O_RDONLY|O_CLOEXEC) = 4
newfstatat(4, "", {st_mode=S_IFREG|0644, st_size=2996, ...}, AT_EMPTY_PATH) = 0
read(4, "# Locale name alias data base.\n#"..., 4096) = 2996
read(4, "", 4096)                       = 0
close(4)                                = 0
openat(AT_FDCWD, "/usr/share/locale/de_DE.UTF-8/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de_DE.utf8/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de_DE/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de.UTF-8/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de.utf8/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de/LC_MESSAGES/bash.mo", O_RDONLY) = 4
newfstatat(4, "", {st_mode=S_IFREG|0644, st_size=125828, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 125828, PROT_READ, MAP_PRIVATE, 4, 0) = 0x7f69bf3f8000
close(4)                                = 0
newfstatat(2, "", {st_mode=S_IFCHR|0600, st_rdev=makedev(0x88, 0), ...}, AT_EMPTY_PATH) = 0
openat(AT_FDCWD, "/usr/share/locale/de_DE.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de_DE.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de_DE/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/usr/share/locale/de/LC_MESSAGES/libc.mo", O_RDONLY) = 4
newfstatat(4, "", {st_mode=S_IFREG|0644, st_size=150826, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 150826, PROT_READ, MAP_PRIVATE, 4, 0) = 0x7f69bf1bc000
close(4)                                = 0
write(2, "/tmp/.mount_FreeCA0MjAB4/AppRun:"..., 116/tmp/.mount_FreeCA0MjAB4/AppRun: Zeile 41:  4509 Speicherzugriffsfehler  (Speicherabzug geschrieben) "${MAIN}" "$@"
) = 116
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_DUMPED, si_pid=4509, si_uid=1000, si_status=SIGSEGV, si_utime=1232 /* 12.32 s */, si_stime=114 /* 1.14 s */} ---
wait4(-1, 0x7fffbea87310, WNOHANG, NULL) = -1 ECHILD (Keine Kind-Prozesse)
rt_sigreturn({mask=[]})                 = 0
read(255, "", 1425)                     = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
exit_group(139)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-24:
@yorikvanhavre can you confirm this?

luzpaz, 2024-10-24:
@semhustej would you be willing to test this ticket ?

kaiwas, 2024-10-24:
I opened the file and inserted 9 windows and one door. I don't see any crash.
[image]

semhustej, 2024-10-24:
I can not reproduce the issue neither. I inserted ca. 20 windows on walls and slanted part of roof too.

- I tried both with RC2 and latest weekly build

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

hds4, 2024-10-25:
Just tested a second Linux machine.
Debian Bookworm 12.5.
FC crashes after inserting six windows.

So it seems it is somewhat OS related ?
What can i do start FC in a sort of "debug" mode that would print some more meaningful messages than strace ?

I will test a Windows 10 machine now.

hds4, 2024-10-25:
Win 10 Pro 22H2 OS build 19045.5011  ,    FC 1.0.0RC2 build 19045
Confirm _no_ crash on Win 10 after 30 windows,

but 3D View stalls with Unhandled Base::Exeception caught in GUIApplication::notify. The error message is: Access violation.
After that i need to close and reopen the house file. Opening a new 3D pane does not stall when the house view stalls.
But that might be a different story.

hds4, 2024-10-25:
Tested now on a completly different machine with freshly installed Kubuntu 22.04.03 LTS. No updates, no additional software installed.
FC 1.0 RC2 crashes after 8 Windows and 1 Door added.

hds4, 2024-10-25:
Tested with Kubuntu 24.04 fresh install. FC 1.0RC2 crashed after one window, but i got this dump from the command line.
Maybe it helps.
----------------------------------------------------------
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Migrating Start Workbench to Start command... done.
Einheitensystem für aktives Dokument aktualisiert
Wähle eine Fläche auf einem existierenden Objekt oder wähle eine Voreinstellung
Wähle eine Fläche auf einem existierenden Objekt oder wähle eine Voreinstellung
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45250) [0x70d81d045250]
#1  0x70d820753128 in Gui::GestureNavigationStyle::EventQueue::forwardAll() from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0xa8
#2  /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so(+0x9591ac) [0x70d8207591ac]
#3  /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so(+0x959599) [0x70d820759599]
#4  0x70d820757962 in Gui::GestureNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0x2a2
#5  0x70d82074a275 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0x65
#6  0x70d820773c1b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0x6b
#7  0x70d820729af5 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0x1a5
#8  0x70d81d6b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Core.so.5+0x7b
#9  0x70d81e57039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#10  0x70d81e5778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#11  0x70d820369079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0x99
#12  0x70d81d6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Core.so.5+0x108
#13  0x70d81e5763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#14  /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x70d81e5c6e6f]
#15  /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x70d81e5ca0c9]
#16  0x70d81e5703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#17  0x70d820369079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0x99
#18  0x70d81d6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Core.so.5+0x108
#19  0x70d81dd353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Gui.so.5+0x84c
#20  0x70d81dd0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Gui.so.5+0xac
#21  /tmp/.mount_FreeCAWEZT6J/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x70d817f15bea]
#22  /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x70d81bd10ad7]
#23  /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x70d81bd13ac7]
#24  /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x70d81bd140ce]
#25  0x70d81d70aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Core.so.5+0x66
#26  0x70d81d6b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Core.so.5+0x11b
#27  0x70d81d6bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libQt5Core.so.5+0x82
#28  0x70d8202c84e1 in Gui::Application::runApplication() from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0xa41
#29  /tmp/.mount_FreeCAWEZT6J/usr/bin/freecad(+0x789f) [0x6026fceb189f]
#30  /lib/x86_64-linux-gnu/libc.so.6(+0x2a3b8) [0x70d81d02a3b8]
#31  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x70d81d02a47b]
#32  /tmp/.mount_FreeCAWEZT6J/usr/bin/freecad(+0x7c12) [0x6026fceb1c12]

hds4, 2024-10-25:
FC weekly build on Debian 12.5 Bookworm does not crash after 40+ Windows added and removed on various Faces.

OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

xtemp09, 2024-10-26:
```cpp
#1  0x70d820753128 in Gui::GestureNavigationStyle::EventQueue::forwardAll() from /tmp/.mount_FreeCAWEZT6J/usr/bin/../lib/libFreeCADGui.so+0xa8
```

I blame the gesture navigation style. @hds4, what do you use? Can you reproduce the crash with another navigation style?

----

I would recommend the moderators to mark this issue as blocker and solve it before the release or disable the gesture navigation style at all if it is found to be the culprit of these crashes.

hds4, 2024-10-26:
I do use Gesture. Did the following tests:
1) Using Gesture. FC 1.0 RC2 crashes immediatly after the first windows.
2) Using TinkerCad. FC does not crash even after tens of windows adding and deleting.
3) Using CAD FC doesn not crash. Same test as 2)
@xtemp09 Good finding !

semhustej, 2024-10-26:
I tried quick test on Windows machine with gesture setup. Inserted ca. 10 windows with no crash. So this might be limited to Linux.

luzpaz, 2024-10-26:
@paullee0 do you mind trying to reproduce ?

paullee0, 2024-10-26:
Hmmm... I have difficulty opening the file :  
- Right click the file, save as in a folder, remove *.zip, and open it, right?
- Not sure why there is error ?

```
00:14:36  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Invalid project file /home/paullee/Downloads/FreeCAD_Models/hds4/HausTest.FCStd: basic_ios::clear: iostream error

```

[image]

xtemp09, 2024-10-26:
@paullee0, I renamed the file into *gz and gunzipped it.

paullee0, 2024-10-26:
There are some problem with FC that there is crash now and then but seems not this one at the moment.

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

[image]

hds4, 2024-10-26:
Hm, from your screenshot: Could it be that you are using CAD navigation style and not Gesture ?

paullee0, 2024-10-27:
> Hm, from your screenshot: Could it be that you are using CAD navigation style and not Gesture ?

I experience random crash in other cases not with Gesture.

I test Gesture with AppImage and crash on one or two Window creation :-

```
paullee@f40-i3-gen12:~/Downloads/FreeCAD/FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311/squashfs-root$ ./AppRun &
[1] 39313
paullee@f40-i3-gen12:~/Downloads/FreeCAD/FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311/squashfs-root$ FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Choose a face on an existing object or select a preset
Program received signal SIGSEGV, Segmentation fault.
./AppRun: line 41: 39320 Segmentation fault      (core dumped) "${MAIN}" "$@"

[1]+  Exit 139                ./AppRun
```

```

OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

hds4, 2024-11-18:
Can confirm FC RC4 with Gesture does not crash after 25 Windows.

OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.51

[image]

closed 2024-11-18 by maxwxyz