FreeCAD/FreeCAD #17443: Resolver Failure on 1.0RC2

OnlineDynamic, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi

I have a sketch design which has lots of 12.4mm holes which are constrained onto the intersection points of a complex pattren of circle construction lines

I'm hitting an issue with the sketch resolver failing in 1.0 RC2... the console is showing these types of errors:

08:41:10  BodySpiralPixels: Solving the sketch failed
08:41:51  BodySpiralPixels: Solving the sketch failed
08:42:01  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (24, 0)
08:42:01  freecad_issue#BodySpiralPixels: App.getDocument('freecad_issue').getObject('Sketch004').movePoint(24,0,App.Vector(-70.445410,-37.429231,0),1)
08:42:01  ViewProviderSketch: Drag curve: Not able to move point with the id and type: (24, 0)
08:42:04  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (24, 0)
08:42:04  freecad_issue#BodySpiralPixels: App.getDocument('freecad_issue').getObject('Sketch004').movePoint(24,0,App.Vector(-96.690934,-54.430351,0),1)
08:42:04  ViewProviderSketch: Drag curve: Not able to move point with the id and type: (24, 0)
08:42:16  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (24, 2)
08:42:16  freecad_issue#BodySpiralPixels: App.getDocument('freecad_issue').getObject('Sketch004').movePoint(24,2,App.Vector(503.238068,-18.162346,0),0)
08:42:16  ViewProviderSketch: Drag point: Not able to move point with the id and type: (24, 2)
08:42:19  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (24, 2)
08:42:19  freecad_issue#BodySpiralPixels: App.getDocument('freecad_issue').getObject('Sketch004').movePoint(24,2,App.Vector(425.907745,-114.337921,0),0)
08:42:19  ViewProviderSketch: Drag point: Not able to move point with the id and type: (24, 2)

I have attached a zipped up copy of the file having the issues - could you please try to load this up and see if the failures are in my design/file creation or whether this is a bug in the resolver

[freecad_issue.zip](https://github.com/user-attachments/files/17503094/freecad_issue.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.48
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-24:
Does the same file work in the stable 0.21.2 version? Can you also try it in the latest weekly builds (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)?
When is the error happening, can you post steps to reproduce?

OnlineDynamic, 2024-10-24:
Not working on 0.21.2 but that might be because it is using Variable Sets - which don't look to be supported in that version.  Error message on opening in 0.21.2 says:

10:57:10  Cannot create object 'VarSet': ('App::VarSet' is not a document object type)
10:57:10  No extension found of type 'Gui::ViewProviderFaceTexture'
10:57:10  No extension found of type 'Gui::ViewProviderFaceTexture'
10:57:10  No extension found of type 'Gui::ViewProviderFaceTexture'
10:57:10  Sketcher constraint number 13 is malformed!
10:57:10  The Sketch has malformed constraints!

I have just tried the weekly build from 2 days ago and still get solver failed to converge in the solver messages and console out says this when trying to move circle named '180-Circle':

11:04:40  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (179, 0)
11:04:40  freecad_issue#BodySpiralPixels: App.getDocument('freecad_issue').getObject('Sketch004').movePoint(179,0,App.Vector(-194.964661,-1052.385620,0),0)
11:04:40  ViewProviderSketch: Drag curve: Not able to move point with the id and type: (179, 0)

OnlineDynamic, 2024-10-24:
@maxwxyz - in terms of steps to reproduce just open the file and you will see the resolver fails on the sketch labelled 'BodySpiralPixels'

JohnOCFII, 2024-10-24:
Tested on weekly and observed the same solver failure.

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.50
```
[image]

maxwxyz, 2024-10-24:
@AjinkyaDahale could you take a look at this?
Can it be related to the file?

Jarno-de-Wit, 2024-10-24:
It wouldn't entirely surprise me if the cause was inside the file. The solver failure seems to be caused by the bottom circle of the "8". More specifically, removing line 6 (the lower of the bottom left radially outwards pointing rays) allows the solver to converge properly, albeit to a underconstrained system since the original file did not have all other things constrained. From this point, I went on the apply the missing constraints, which gave me the resulting properly converging file:
[freecad_issue.FCStd.zip](https://github.com/user-attachments/files/17510561/freecad_issue.FCStd.zip)
(Note: First, I separated the drawing into two separate sketches, and on the top section, I replaced some diameter constraints with equality for testing. Later on, I recombined the two sketches using the carbon copy function.)

FlachyJoe, 2024-10-24:
FreeCAD offers functions for circular array, in Draft or PartDesign workbenches. Please use them!

OnlineDynamic, 2024-10-24:
> FreeCAD offers functions for circular array, in Draft or PartDesign workbenches. Please use them!

@FlachyJoe I do for more simplistic use cases but they don't look to have the capabilities to cover inter-secting circular arrays - the file shared is just part of something much more complex - if it is do'able I'm all ears! and open to see how you would address creating the pattern shared

FlachyJoe, 2024-10-24:
@OnlineDynamic please go to the forum, feel free to mention me in order I see your message.

OnlineDynamic, 2024-10-24:
@FlachyJoe will do, that aside I think the resolver failing and not giving an output for the op to resolve failure mode still needs looking into

Jarno-de-Wit, 2024-10-24:
A bit more investigation shows that the responsible constraint is constraint 512, which is a point-on-line constraint for the outer endpoint of line 7, and the main body of line 6. Since both line 6 and line 7 have a coincident constraint to the centre of the circle, and a constrained direction, the only way this constraint could be adhered to, is if line 7 gets a length of 0. I assume this somehow trips up the solver, making it unable to find a proper solution to the system. In this case, the system is not theoretically overconstrained (a valid solution exists if the length == 0), but numerically, it might very well be.

This is probably not helped by the fact that the two endpoints are not directly coincident, but instead are connected through other coincident constraints / points. Additionally, the angles are defined relative to another line, instead of from an axis. Furthermore, the system had less constraints than there were degrees of freedom, making it harder to quickly detect overconstrainedness. Perhaps together these aspects give the solver just enough wiggle room to think there might be a solution, without it being able to find one due to e.g. numerical noise.

I was able to reproduce the non-convergence in the most recent weekly build (39024) using as few as 3 lines and 1 point, as done in the file below. 
[NonConvergence.FCStd.zip](https://github.com/user-attachments/files/17513391/NonConvergence.FCStd.zip)

Regardless, I think this is a rather unconventional case, considering the issue is caused by the solver needing to create a line with 0 length, thus this shouldn't be a high priority issue.

FlachyJoe, 2024-10-25:
Notice sketcher can construct a null length segment but solver complain:
```
13:28:36  Updating geometry: Error build geometry(2): Both points are equal
13:28:36  Invalid solution from DogLeg solver.
13:28:36  Updating geometry: Error build geometry(2): Both points are equal
13:28:36  Invalid solution from LevenbergMarquardt solver.
13:28:36  Important: the BFGS solver succeeded where the DogLeg and LevenbergMarquardt solvers have failed.
13:28:36  Sketcher::Solve()-BFGS-T:0.00062393
```
Set angle constraint to 180° to create 2 null edges.
[image]

Jarno-de-Wit, 2024-10-25:
While it sometimes works, even your example doesn't always converge. If you set the radius of the circle to 2000 mm, the solver will once again fail to converge. Similarly, moving the centre to 2000 mm x 2000 mm, and constraining the top point using another vertical construction line to the centre makes the solver struggle to converge on the first iteration (though it does seem to converge if more recomputes are triggered manually).
On a similar note, if in my example file the centre point is constrained to coincide with the origin of the sketch, the solver also converges just fine.

This to me seems a suspicous lot like the solver struggling with the floating point tolerances, with your case being a bit more resilient to this numerical noise due to e.g. smoother gradients for the solver to descend, but definitely not completely resistent.