FreeCAD/FreeCAD #17445: Model/Tree View has disappeared

fester225, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I started using 1.0.0RC2. At some point I managed to lose the Model/Tree View. I have found no way to get it back. 'HELP'  appears to be solely for help with Mac devices.
How do I get the Model/Tree View back?

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fester225, 2024-10-24:
Go to VIEW/PANELS, click on MODELS.

maxwxyz, 2024-10-24:
And use the forum for help request.

closed 2024-10-24 by maxwxyz