FreeCAD/FreeCAD #17447: Segfault in Sketcher with CenterOfMass macro

jes, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm running FreeCAD 0.21.2 on Ubuntu 24.04 from the Snap store.

I found that with a particular file (in the attached zip file [sovereign-scales4.zip](https://github.com/user-attachments/files/17504459/sovereign-scales4.zip) ), I can reliably trigger a segfault as follows:

 * delete the existing CenterOfMass item in the tree
 * make a new CenterOfMass with both of the bodies
 * click "Update" and "Save" and tick the eye make the centre of mass visible
 * open up the first sketch in the first body and delete a couple of elements
 * click the "Line" icon in the toolbar

And then FreeCAD instantly segfaults. I did try to simplify this by skipping all of the steps to do with the CenterOfMass macro but could not reproduce the problem without using CenterOfMass. I haven't exhaustively searched to find out whether every single click is necessary, but hopefully this is useful enough as a bug report.

I have a screen recording demonstrating the problem here: https://www.youtube.com/watch?v=VMFdUpOoCko

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

This is the log from the terminal, including a backtrace: [100135.txt](https://github.com/user-attachments/files/17504407/100135.txt)

Apart from the segfault there appears to be a use of deprecated implicit type conversion in CenterOfMass:

```
/home/jes/snap/freecad/common/CenterOfMass.FCMacro:307: DeprecationWarning: an integer is required (got type PySide2.QtCore.Qt.AlignmentFlag).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
  label_comboUnit.setAlignment(QtCore.Qt.AlignRight | QtCore.Qt.AlignVCenter)
```

Also, it didn't happen this time, but in other clicking around trying to reproduce the problem, I saw this error in the terminal, but with no associated segfault:

```
Traceback (most recent call last):
  File "/home/jes/snap/freecad/common/CenterOfMass.FCMacro", line 551, in on_focusChanged
    if w is not None and w.parent() == self.solidLayout.parentWidget():
RuntimeError: Internal C++ object (PySide2.QtWidgets.QWidget) already deleted.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jes, 2024-10-24:
I tried this again in the "latest/edge" version from Snap store:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39036 (Git) Snap 1171
Build type: Release
Branch: main
Hash: 2d57e9f2afa1da442a20a3426719055fc4e906a9
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Fusion
```

And I so far haven't manage to reproduce the segfault in the new version, but I am getting this in the "Report view", so perhaps the segfault is fixed but the CenterOfMass bug that triggered it still remains?:

```
10:21:04  Traceback (most recent call last):
10:21:04    File "/home/jes/snap/freecad/common/CenterOfMass.FCMacro", line 551, in on_focusChanged
10:21:04      if w is not None and w.parent() == self.solidLayout.parentWidget():
10:21:04  RuntimeError: Internal C++ object (PySide2.QtWidgets.QGraphicsView) already deleted.
```

maxwxyz, 2024-10-24:
Please open the issue in the macro repo: https://github.com/FreeCAD/FreeCAD-macros
You can copy your findings there.

closed 2024-10-24 by maxwxyz