FreeCAD/FreeCAD #17451: CAM The mm/min function is no longer available

Roberto8303, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Why is it no longer possible to set the display of movements and speed to mm/min but only mm/sec?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: Dark-blue.qss/FreeCAD Classic/Qt default
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * FreeCAD-themes 2024.7.24
  * Glass
  * IconThemes
  * OpenTheme 2024.9.1
  * PieMenu 1.8.1
  * Render 2024.9.25
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-24:
Is this a regression ?

Roberto8303, 2024-10-24:
Yes, in the old versions I could set it now not

luzpaz, 2024-10-24:
@Roberto8303 can you screenshot or mention the workflow you used previously to enable this feature ?

Syres916, 2024-10-31:
@Roberto8303 the Units Schema which 'chooses' whether it be _mm/s_ or _mm/min_ to be shown has been enhanced during the last development cycle. Previously it was a Global setting applied to any document created or opened, now there is a `System` wide setting for new documents in _Preferences > General > Default unit system_ and the `Document` level setting which is saved with the doc, this is set using the push button in the bottom right hand corner as per:

[image]

Syres916, 2024-10-31:
The enhancement is listed in the [Release Notes](https://wiki.freecad.org/Release_notes_1.0#User_interface) (find under **Further user interface improvements**) , though I can see why it maybe missed in the large list of changes

luzpaz, 2024-10-31:
Thanks syres. Closing.

closed 2024-10-31 by luzpaz