FreeCAD/FreeCAD #17452: PartDesign: Unexpected and unwanted behaviour using fillets

CJVienna, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Misbehaviour occurs even with simple objects (probably the same cause, but different manifestations).
Samples:
#1: Start with a padded rectangle ... select any edge ... make a fillet ... ok ... select an adjacent edge ... make a fillet ... oops, all three edges (those with the same vertex in common) become rounded. Unfortunately I can't remove the unwanted one later on.
#2: Start with a padded rectangle ... select any face ... make a fillet ... ok ... select any adjacent edge ... make a fillet ... oops, "Fillet001: BRep_API: command not done" --> "Error!" "Body: Tip shape is empty"

Unsatisfying workaround: Select all adjacent faces / edges where you want fillets applied and make fillet ... ok, but with the disadvanage you can't apply corrections later on without sometimes breaking referenced objects

Chamfer as a similar tool does not show those strange behaviours

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * ArchTextures
  * Assembly4 0.11.12

same behaviour as in previous version 0.21.2:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Austria (de_AT)
Installed mods: 
  * ArchTextures
  * Assembly4 0.11.12
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-24:
@CJVienna did you cross-ref preexisting fillet bugs before opening this one? https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+label%3A%22Topic%3A+Fillets%22

CJVienna, 2024-10-25:
@luzpaz yes, of course, I checked the preexisting fillet bugs. (and now I did again).

Really? An example file request for a simple pad?
Ok, there are two sample files attached in a ZIP
[Errors.zip](https://github.com/user-attachments/files/17525642/Errors.zip):
**Fillet_1** with Body.Pad.Edge4 selected and make Fillet executed. If you try to make an additional fillet on Body.Fillet.Edge4 then Body.Fillet.Edge13 unintendedly gets also a fillet.
**Fillet_2** with Body.Pad.Face6 selected and make Fillet executed. If you try to make an additional fillet on Body.Fillet.Edge4 then the error described in the problem description occurs.

B.t.w: I tried to also create a macro, but macro execution fails in V1.0 with <class 'NameError'>: name 'ActiveSketch' is not defined. But that is a different story.

luzpaz, 2024-10-25:
> B.t.w: I tried to also create a macro, but macro execution fails in V1.0 with <class 'NameError'>: name 'ActiveSketch' is not defined. But that is a different story.

Thanks, do you mind opening ticket for that? TIA

CJVienna, 2024-10-25:
> > B.t.w: I tried to also create a macro, but macro execution fails in V1.0 with <class 'NameError'>: name 'ActiveSketch' is not defined. But that is a different story.
> 
> Thanks, do you mind opening ticket for that? TIA

so I did (#17502)

luzpaz, 2026-09-04:
@pieterhijma is this OCC related ?