FreeCAD/FreeCAD #17457: CAM Path feeds and speeds no longer visible

Roberto8303, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
In version 0.21.1 the function in addon manager feeds and speeds works regularly

Nella versione 1.1.0dev una volta scaricata ed installata non e’ visibile e non può essere utilizzata 
[image]

[image]

### Full version info

```shell
Versione 1.1.0dev 39024git
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-24:
@Roberto8303 full [About](https://wiki.freecad.org/About) info please, TIA! (edit the OP and add it please)

Roberto8303, 2024-10-24:
What should I do I don't understand, on this version the installed addon is not seen in the CAM environment, it only works on PATH in versions 0.21.1
No errors are reported

luzpaz, 2024-10-24:
Please read carefully: 
I mentioned above that we need your Full About info and provided you with a link: https://wiki.freecad.org/About  
This link instructs you to go in to: Help > About FreeCAD > `Copy to clipboard`  
then you come back to this ticket and  paste your full version info. 
Result: In it will tell us what addons you have installed along with other information.

It sounds like there is an issue with a 3rd party addon Feeds and Speeds (https://github.com/dubstar-04/FeedsAndSpeeds). I opened a ticket there asking if the addon was ready for 1.0 (https://github.com/dubstar-04/FeedsAndSpeeds/issues/56) but still haven't heard back from @dubstar-04

Roberto8303, 2024-10-24:
[image]
[image]

Roberto8303, 2024-10-24:
[image]

Roberto8303, 2024-10-24:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * FreeCAD-themes 2024.7.24
  * Glass
  * IconThemes
  * OpenTheme 2024.9.1
  * PieMenu 1.8.1
  * Render 2024.9.25

luzpaz, 2024-10-24:
This is an issue with 3rd-Party Addon. The ticket needs to be opened at https://github.com/dubstar-04/FeedsAndSpeeds

CC @chennes no response from @dubstar-04 so far, perhaps we should deprecate the addon for 1.0 to avoid influx of tickets like this?

dubstar-04, 2024-12-12:
Addon updated to support FreeCAD 1.0