FreeCAD/FreeCAD #17458: Sketcher and Part Design: Several serious issues involving dashed blue lines on Ubuntu

FEA-eng, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I don't know if it's all related otherwise than involving dashed blue lines but there's a whole bunch of serious issues on my Ubuntu VM using clean config and the newest AppImage:

1. The way the global origin planes are displayed:

[image]

2. The way the Sketcher selection box appears:

[image]

3. The worst one - geometry getting completely broken after applying the construction mode:

[image]

Is it only on my side? Some graphics glitch? I don't observe such issues on Windows. Can it be related to #17028 in any way?

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-24:
You've mentioned it is happening in the latest AppImage version - is it normal in the RCs or the stable version on Ubuntu?

kaiwas, 2024-10-25:
there is a possibility that some display problems may be related to the "drivers" of the virtual machine's video adapter. I have encountered similar difficulties in other programs.

I don't have Ubuntu installed, but I can check this issue in Manjaro

kaiwas, 2024-10-25:
1.
[image]
2.
[image]
3.
[image]
```

OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * lattice2 1.0.0
  * PitchedRoof
  * addFC 0.7.4
  * OpenTheme 2024.9.1
```

FEA-eng, 2024-10-25:
> You've mentioned it is happening in the latest AppImage version - is it normal in the RCs or the stable version on Ubuntu?

RC2 has the same issues. Stable too, apart from the construction geometry one.

maxwxyz, 2025-08-21:
Which of these? The origins got reworked and there is a PR for the sketcher box selection: https://github.com/FreeCAD/FreeCAD/pull/23261