FreeCAD/FreeCAD #17459: BIM: IFCSite Ref Elevation exported in wrong Unit

Grosskopf, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Thank you for fixing my other Issue so fast, I hope I'm not bothering you too much, but I found one more:

I have a Project with a Ref Elevation set to 144m
[image]

The export mentions these two lines:
```
#13=IFCSIUNIT(*,.LENGTHUNIT.,$,.METRE.);
```
```
#24=IFCSITE('0uDcgZkBX9vhe3V9krnvua',#5,'Site','',$,$,$,'Site',.COMPLEX.,(0,0,0),(0,0,0),144000.,$,$);
```
so I think the Site is now at 144000m

I was a bit perplexed by this, and I checked the IFC docs just to be extra sure, but I believe this Bug shot my Site to Space :D

while yes, in the Definition of those IfcLengthMeasures mentions that they are "usually in Millimeters", looking at the examples from buildingsmart reveals that other IfcLengthMeasures like the length and height of an IfcDoor do fall under the Influence of this IFCSIUNIT https://standards.buildingsmart.org/IFC/DEV/IFC4_2/FINAL/HTML/annex/annex-e/basic-unit-declaration.htm

the complete IFC File exported like this:
```
ISO-10303-21;
HEADER;
FILE_DESCRIPTION(('ViewDefinition [CoordinationView]'),'2;1');
FILE_NAME('IFC_Export-Site.ifc','2024-10-24T13:09:51',('',''),(''),'IfcOpenShell 0.7.0','IfcOpenShell 0.7.0','');
FILE_SCHEMA(('IFC4'));
ENDSEC;
DATA;
#1=IFCPERSON($,$,'',$,$,$,$,$);
#2=IFCORGANIZATION($,'',$,$,$);
#3=IFCPERSONANDORGANIZATION(#1,#2,$);
#4=IFCAPPLICATION(#2,'1.0 build 0','FreeCAD','118df2cf_ed21_438e_a41');
#5=IFCOWNERHISTORY(#3,#4,$,.ADDED.,1729775391,#3,#4,1729775391);
#6=IFCDIRECTION((1.,0.,0.));
#7=IFCDIRECTION((0.,0.,1.));
#8=IFCCARTESIANPOINT((0.,0.,0.));
#9=IFCAXIS2PLACEMENT3D(#8,#7,#6);
#10=IFCDIRECTION((0.,1.,0.));
#12=IFCDIMENSIONALEXPONENTS(0,0,0,0,0,0,0);
#13=IFCSIUNIT(*,.LENGTHUNIT.,$,.METRE.);
#14=IFCSIUNIT(*,.AREAUNIT.,$,.SQUARE_METRE.);
#15=IFCSIUNIT(*,.VOLUMEUNIT.,$,.CUBIC_METRE.);
#16=IFCSIUNIT(*,.PLANEANGLEUNIT.,$,.RADIAN.);
#17=IFCMEASUREWITHUNIT(IFCPLANEANGLEMEASURE(0.017453292519943295),#16);
#18=IFCCONVERSIONBASEDUNIT(#12,.PLANEANGLEUNIT.,'DEGREE',#17);
#19=IFCUNITASSIGNMENT((#13,#14,#15,#18));
#20=IFCDIRECTION((6.12323399573677E-17,1.));
#21=IFCGEOMETRICREPRESENTATIONCONTEXT($,'Model',3,1.E-05,#9,#20);
#22=IFCGEOMETRICREPRESENTATIONSUBCONTEXT('Body','Model',*,*,*,*,#21,$,.MODEL_VIEW.,$);
#23=IFCPROJECT('3tVJDgeS54XeF5rwsRmcH5',#5,'Boolean_IFC_Export',$,$,$,$,(#21),#19);
#24=IFCSITE('0uDcgZkBX9vhe3V9krnvua',#5,'Site','',$,$,$,'Site',.COMPLEX.,(0,0,0),(0,0,0),144000.,$,$);
#25=IFCPROPERTYSINGLEVALUE('MyProperty',$,IFCBOOLEAN(.F.),$);
#26=IFCPROPERTYSET('0DVxYVBTP7gwWVZLhXKj0A',#5,'custom',$,(#25));
#27=IFCRELDEFINESBYPROPERTIES('377qOvip94we5uNjbJO5Ta',#5,$,$,(#24),#26);
#28=IFCRELAGGREGATES('2Nnzpmm9PA$B9cEhZcVjl6',#5,'ProjectLink','',#23,(#24));
#29=IFCBUILDING('1An0InaCjD4hyQSrFepdCK',#5,'Default Building','',$,$,$,$,.ELEMENT.,$,$,$);
#30=IFCRELAGGREGATES('0uBxeHFVP64RlaCGLonK6p',#5,'SiteLink','',#24,(#29));
ENDSEC;
END-ISO-10303-21;

```

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-24:
> I hope I'm not bothering you too much, but I found one more   

Not at all. This is exactly the feedback we are looking for. Keep reporting bugs! :pray:

semhustej, 2024-10-24:
I can confirm the issue. 

Steps to reproduce:

- Open sample file - site is at level 144m 
[FC-site-elevation.zip](https://github.com/user-attachments/files/17508679/FC-site-elevation.zip)

- Export to IFC
- Site is at 144.000m in the IFC file

[image]

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

semhustej, 2024-10-24:
Related note: When my test FCstd file is converted to strict IFC mode and than saved (Ctrl+S), the eleveation is set to correct level 144.000mm. See the difference in units of the IFC file.

Export as IFC:
[image]

Switch to strict IFC and save:
[image]

I am not sure how setting both .mili. and .metre. in IFC file works. I checked BonsaiBIM and when project is set to meters as a unit, the IFC contains just .metre. as a unit. I would suppose strict IFC should also use just one.

Grosskopf, 2024-10-24:
.milli. after the second comma and .metre. after the third is perfect

as per Documentation from buildingsmart at https://standards.buildingsmart.org/IFC/DEV/IFC4_2/FINAL/HTML/schema/ifcmeasureresource/lexical/ifcsiunit.htm

the third Parameter in an IfcSIUnit is an IfcSIPrefix, one of those listed here: https://standards.buildingsmart.org/IFC/DEV/IFC4_2/FINAL/HTML/schema/ifcmeasureresource/lexical/ifcsiprefix.htm among those, milli is listed :)

I ran across an IFC File with a millimetre ifcsiunit before from ArchiCAD, Revit also usually exports in full meters but since I found this IfcSIUnit in ArchiCAD I had to implement looking for this IfcSIUnit for my usecases before :D

semhustej, 2024-10-24:
> the third Parameter in an IfcSIUnit is an IfcSIPrefix, one of those listed here: https://standards.buildingsmart.org/IFC/DEV/IFC4_2/FINAL/HTML/schema/ifcmeasureresource/lexical/ifcsiprefix.htm among those, milli is listed (with the wrong exponent though, whoops, found another bug somewhere else xD) :)

I thought .mili.,.meter. means somehow both milimeters and meters in one document :). Thanks for clarifying that .mili. is a prefix. Now it makes sense.

hoshengwei, 2024-10-25:
From my understanding, FreeCAD's IFC export uses meters as the default unit, meaning all lengths are converted to meters when exported to an IFC file. I'm not sure if the new Lock System supports using other units yet.

It's not that IFC doesn't support other length units; rather, it's just more convenient to standardize on meters in the IFC file, as it simplifies coding by removing the need for an additional layer of unit conversion.

When reading an IFC file, regardless of the default length unit used in the file, FreeCAD utilizes ifc_tools.get_scale() to convert all lengths in the IFC file into FreeCAD's base unit, millimeters, which are then displayed in the user’s preferred length unit (e.g., centimeters in EuroBuilding).

So, regardless of the unit used, the measurements should convert to the length unit you've set in FreeCAD and display correctly.

Let me know if I've misunderstood anything!

yorikvanhavre, 2024-10-29:
> I'm not sure if the new Lock System supports using other units yet.

Yes it does, the new NativeIFC system uses a better scaling system that depends the file itself. The old, Arch IFC exporter, that is still used by the NativeIFC, takes whatever scaling value you give it, only when you export an IFC file from scratch, a default metric template is built for you.

Your PR is an audacious idea, but let's try :)

PR #17492 (2024-10-29): [BIM] Fix IFC export elevation with wrong unit
Fix #17459 

in origin code
there is only "ElevationWithFlooring"  "Elevation" two attributes will run multiple the scale
however in IFC Schema have other kind of Elevation
eg:  IfcSite has RefElevation
eg: IfcBuilding has ElevationOfTerrain & ElevationOfRefHeight
there might be other else

but I think all of them will including the "Elevation" string in the attributes name
so I think this fix might work
closed 2024-10-29 by yorikvanhavre