FreeCAD/FreeCAD #17461: Draft: Deleting an object while editing crashes Draft workbench

m1cha1, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Step to reproduce:
1. Create rectangle
2. Edit rectangle
3. Select rectangle during editing and press delete

Result:
Editing points are still on the screen.
Once you've finished editing, you can't do anything in the Draft Workbench. The only way to fix this is to restart Freecad.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Draft

### Anything else?

Error message:
16:08:42  pyException: Traceback (most recent call last):
  File "D:\Program Files\Freecad\Mod\Draft\DraftGui.py", line 930, in reject
    self.callback()
  File "D:\Program Files\Freecad\Mod\Draft\DraftGui.py", line 1063, in finish
    self.sourceCmd.finish(cont=cont)
  File "D:\Program Files\Freecad\Mod\Draft\draftguitools\gui_edit.py", line 337, in finish
    self.deformat_objects_after_editing(self.edited_objects)
  File "D:\Program Files\Freecad\Mod\Draft\draftguitools\gui_edit.py", line 815, in deformat_objects_after_editing
    obj_gui_tools.restore_object_style(obj, self.objs_formats[obj.Name])
                                                              ^^^^^^^^
<class 'RuntimeError'>: This object is currently not part of a document

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-10-24:
I can confirm the issue.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

maxwxyz, 2024-10-24:
@Roy-043 FYI

semhustej, 2024-10-24:
I am also able to reproduce the issue on FC 0.21.2.

Roy-043, 2024-10-25:
I cannot reproduce this. While the Rectangle is in edit mode the Std_Delete command simply does not work. And if I delete the object through Python code, there is no error.

Maybe there is a relation with the mouse navigation style (I use CAD)? Can you both please retest with a 'fresh' CFG? Also please verify if the Rectangle is in fact in edit mode (indicated by a special background color in the Tree view).

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

semhustej, 2024-10-25:
Hi,

this is after .cfg files are removed - default FC setup.  Hope it helps to clarify my steps. I am able to reproduce this 100% times.
I do not see special background color in edit mode though.

https://github.com/user-attachments/assets/4549a02e-a5fd-44fc-983f-c710122a0501

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

Roy-043, 2024-10-25:
Thanks for the helpful video. I can reproduce now. I was double-clicking the object in the Tree view for the "Edit rectangle" step. If you use the Draft_Edit command instead the object is not put into edit mode(*) and can indeed be deleted.

(*) This is probably because only one object at a time can be put into edit mode, and the Draft_Edit command supports selecting multiple objects.

Fixing the mentioned crash is relatively easy, but preventing the delete operation would be better I think. But I am not sure if that is possible.

commit e28c64b6 (2024-10-28): Draft: Fixes #17461
commit 231d734b (2024-10-28): Draft: Fixes #17461
closed 2024-10-28 by yorikvanhavre