FreeCAD/FreeCAD #17463: Limitation when using Polar Pattern to create multi start threads after using thread profile and Subtractive Helix on internal bore.

Bladevane, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Solid walled cyclinder created. Thread profile added as a sketch. Use Subtractrive Helix to create an internal thread. All OK. I needed a 2 start thread so tried to create this using Polar Pattern. Failed unless length of cyclinder was very short. Would not show multi thread profile when length of cyclinder was 10mm, but would when it was 8mm. Pitch of thread was 4 mm.

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.44
  * Curves.backup1724060728.1499815
  * DynamicData 2.62.0
  * DynamicData.backup1724060736.0375865
  * fasteners 0.5.27
  * fasteners.backup1724060744.1188815
  * Help 1.0.3
  * lattice2 1.0.0
  * lattice2.backup1724060752.2744286
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pure-machinery, 2024-10-24:
I did a small test and for me it works.

I've picked some arbitrary threads with 1.5 thread pitch, and made it start 3 times. The length is about 100 mm.

[image]
[image]

[threads_example.zip](https://github.com/user-attachments/files/17511213/threads_example.zip)

mwganson, 2024-10-24:
By the way ThreadProfile can make 2 start and 3 start threads for the v thread profile types.  Look at the Variants property in the ThreadProfile object.

Bladevane, 2024-10-25:
Mark,

 

Your time is appreciated as much as your response. However, because I am only a tinkering amateur I am finding it impossible to follow how to create such an internal thread.

 

Thanks for your input.

 

Glyn.

AIRCAP, 2024-10-25:
an example FCStd file with the failed threads would be useful.

mwganson, 2024-10-25:
I will make a few assumptions.  You have created a solid cylinder by padding a sketch.  Hide the pad, and show the sketch.  You can do this by selecting an object in the tree view and pressing the space bar to toggle its visibility.  Delete the later failed objects, but keep the sketches used to make them in case you decide to use them again.  The only thing visible should be the sketch with the circle used to make the cylinder.

Next step is to go to ThreadProfile workbench and create the standard V thread profile.  Select the new object in the tree view, and then find the MapMode property in the attachment section.  Click the [...] button that comes up when the property is selected to bring up the attachment editor.  Select the circle in the 3d view, ensure Concentric is the attachment mode, and close the dialog with the ok button.

Since we are making internal threads we must set the ThreadProfile InternalOrExternal property to Internal.  Then look for the Presets property (V Thread Presets) and select the desired preset for the thread diameter and pitch.  You mentioned 4 mm pitch, so I will make another assumption that the desired size is M36 x 4 Coarse Thread, but there are other options and you can also set the custom pitch and minor diameters without using the presets.  With Internal set, with Preset set, now we can set the Variants property to 2 start option.  Other options: 60 (for standard 60 degree threads), 45 (for 3d printer friendly 45 degree custom threads), 2 start, and 3 start (the 2-start and 3-start options are both 60 degree).  Final property to set is the height.  This is 10 mm by default, but if your pad is 50 mm, then make this at least 50 to go all the way through unless you want a blind hole. 

With all the ThreadProfile object's properties set we are now ready to make the helix that it will use to do the sweep.  Select the threadprofile object and click the create helix toolbar icon.  Make note of which direction the helix is pointing into.  If it's the wrong direction, go to the ThreadProfile object properties and toggle the MapReversed property from false to true.  This will put the helix on the opposite side facing the other direction.  With both the helix and the VThreadProfile object selected, click the sweep toolbar icon to do the subtractive pipe.

You can go back in and edit the VThreadProfile properties and everything should automatically recompute if you want to make some tweaks to the parameters.  If you require additional help with ThreadProfile workbench create a new topic on the help board and mention me [mention]TheMarkster[/mention] so we're not polluting everybody's e-mail with this discussion.

Bladevane, 2024-10-25:
Mark,

 

That help works like a charm. Thank you so much for the detailed “what -to-do” steps. I created a 2 and 3 start thread by following your instructions. Amazing.

 

Thank you again,

 

Glyn.

Bladevane, 2024-10-25:
Eric,

 

I have received and followed a detailed step by step procedure from Mark Ganson and it works like a dream. Thank you for your prompt responses, I think I am now all set to proceed with my design project.

 

Regards,

 

Glyn.

maxwxyz, 2024-10-26:
@Bladevane so the issue is resolved?

Bladevane, 2024-10-26:
Max,I say Yes, thanks to the superb instructions given to me by Mark Ganson and my realisation that I was probably going about it the wrong way. So all good. Regards,Glyn.

closed 2024-10-26 by maxwxyz