FreeCAD/FreeCAD #17464: CAM_Sanity reduces the window of the drawing

duub, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After clicking "Check the CAM job for common errors" icon, it asks to save and html file with some information and after that i get the drawing window reduced in size, screenshot:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: Spanish/Spain (es_ES) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.29
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-24:
Can you try it in the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

duub, 2024-10-24:
Done. Still happens the same issue, but now I'm not getting that error on the report view.

Syres916, 2024-10-30:
Not only confirmed but it's worse, in that if the user had _Orthographic_ display setting, it's been changed to _Perspective_ and not reverted to how it was before running Sanity and the FPS is going ballistic **_like_** the model is being moved back and forth a very small distance. This is definitely a regression, it used to work correctly a few months ago.

Expore-it-all, 2025-06-13:
Just Commenting here that this is still present on current weekly build as of 06/12/2025

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1

Current screenshot

[image]

Though I do find that changing the window from the current file to another file or the start menu will resolve the windowing issue.