FreeCAD/FreeCAD #17465: Segmentation fault recieved after revolving a fresh sketch

pure-machinery, 2024-10-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Blank document
2. Create a body
3. Create sketch
4. Pick a plane
5. Create a rectangle that lies on the revolution axis
6. Enter revolve command (now here's comes the fun part)
6.a If I directly select the specific axis I can see the revolve succeed but afterwards Freecad _crashes_.
6.b If I pick (depending on the plane) X - axis for revolution it succeeds and _does not crash_. Afterwards if I try and change from X-axis to selected axis it also succeeds and _does not crash_. 

Attaching a file makes no sense since i can not save it after it crashes.

<details><summary>Backtrace</summary>

```c

Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x796a78a4c1d0]
#1  0x796a6a070cfe in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /usr/lib/libTKBRep.so.7.8+0x2e
#2  0x796a6a080c14 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /usr/lib/libTKBRep.so.7.8+0x74
#3  0x796a6a083062 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /usr/lib/libTKBRep.so.7.8+0xe2
#4  0x796a6b4bb4f2 in Measure::Measurement::length() const from /usr/lib/freecad/lib/Measure.so+0x1c2
#5  0x796a534c4265 in MeasureGui::QuickMeasure::printResult() from /usr/lib/freecad/lib/MeasureGui.so+0x475
#6  0x796a534c5655 in MeasureGui::QuickMeasure::processSelection() from /usr/lib/freecad/lib/MeasureGui.so+0x25
#7  /usr/lib/libQt6Core.so.6(+0x1b2229) [0x796a791b2229]
#8  0x796a791bb685 in QTimer::timerEvent(QTimerEvent*) from /usr/lib/libQt6Core.so.6+0xa5
#9  0x796a791a3439 in QObject::event(QEvent*) from /usr/lib/libQt6Core.so.6+0x109
#10  0x796a7a0fe31a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8a
#11  0x796a7c2a7249 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#12  0x796a791585a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x1a8
#13  0x796a792d5d1f in QTimerInfoList::activateTimers() from /usr/lib/libQt6Core.so.6+0x5ef
#14  /usr/lib/libQt6Core.so.6(+0x3c23c1) [0x796a793c23c1]
#15  /usr/lib/libglib-2.0.so.0(+0x5d559) [0x796a777cc559]
#16  /usr/lib/libglib-2.0.so.0(+0xc0157) [0x796a7782f157]
#17  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x796a777cba55]
#18  0x796a793bf71d in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#19  0x796a79164566 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x226
#20  0x796a79159a2f in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x9f
#21  0x796a7c1af4e7 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xd07
#22  freecad(+0x6cbc) [0x61b1cae83cbc]
#23  /usr/lib/libc.so.6(+0x25e08) [0x796a78a34e08]
#24  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x796a78a34ecc]
#25  freecad(+0x70b5) [0x61b1cae840b5]

```

</details>

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * BillOfMaterials 1.0.1
  * sheetmetal 0.5.2
  * fasteners 0.5.29
  * fcgear 1.0.0
  * OpenTheme 2024.9.1
  * IconThemes
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-24:
can you try this in the weekly builds?
@AIRCAP should this be fixed in the weekly?

AIRCAP, 2024-10-24:
@pure-machinery  I can't get the weekly to crash. but maybe I'm doing it wrong, I get an "Revolution: Revolve axis intersects the sketch" error but no crash.  Can you provide a video or more detailed info? What do you mean with  **that lies on the revolution axis** - around it? touching it with a side? constrain a corner to it?

pure-machinery, 2024-10-24:
[Screencast From 2024-10-24 21-56-04.webm](https://github.com/user-attachments/assets/5af01576-f65e-4653-adc3-e89906c740ce)

wwmayer, 2024-10-24:
Fixed with https://github.com/FreeCAD/FreeCAD/pull/16921

luzpaz, 2024-10-25:
@maxwxyz can you trigger weekly so users can test this fix (per wwmayer :point_up_2: )

maxwxyz, 2024-10-26:
Dove

luzpaz, 2024-10-26:
@pure-machinery can you test the weekly to confirm fix ?

pure-machinery, 2024-10-26:
I reproduced the steps, and for now it works as expected and doesn't crash.

Here's the version:

```
OS: Arch Linux (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * BillOfMaterials 1.0.1
  * sheetmetal 0.5.2
  * fasteners 0.5.29
  * fcgear 1.0.0
  * OpenTheme 2024.9.1
  * IconThemes
```

closed 2024-10-26 by luzpaz