FreeCAD/FreeCAD #17472: VarSet: Dialog cannot be fully translated

wwmayer, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The method uses a std::string to create the window title and assigns it to a QString. This way it's impossible for the i18n tools to pick up the string literal.

https://github.com/FreeCAD/FreeCAD/blob/499d192393d6ba71a7b04b0a0813a2c4789126c6/src/Gui/DlgAddPropertyVarSet.cpp#L174

The part `""Add a property to"` must be moved into `tr()`. And in order for the i18n tools to get the correct context this liine must be added (before the constructor): `/* TRANSLATOR Gui::Dialog::DlgAddPropertyVarSet */`

### Full version info

```shell
FreeCAD 1.1.0, Libs: 1.1.0devR39035 (Git)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-25:
@pieterhijma FYI

wwmayer, 2024-10-25:
Another issue is that the dialog doesn't react on language changes. The dialog is modeless and thus allows to open the preferences. If the user switches to another language the dialog doesn't update.

This can be easily achieved by overriding `void changeEvent(QEvent* e)`:
```
void DlgAddPropertyVarSet::changeEvent(QEvent* e)
{
    if (e->type() == QEvent::LanguageChange) {
        ui->retranslateUi(this);
    }
    QDialog::changeEvent(e);
}
```

pieterhijma, 2024-10-25:
Thanks, I will fix it.

PR #17480 (2024-11-22): Gui: Improve localization support in VarSet dialog
Closes #17472