FreeCAD/FreeCAD #17475: VarSet: The dialog causes a segmentation fault

wwmayer, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With these steps the dialog crashes reliably:
1. New document
2. Open the VarSet dialog
3. Enter a name
4. Activate the option "Add another"
5. Click OK
6. Click Cancel
7. Open the VarSet dialog again
8. Click Cancel
9. => Crash!

### Full version info

```shell
FreeCAD 1.1.0, Libs: 1.1.0devR39035 (Git)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-25:
@pieterhijma FYI

pieterhijma, 2024-10-25:
Looking into it.

pieterhijma, 2024-10-28:
On one computer I could reproduce:

```
OS: Arch Linux (xcb)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38997 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: 146f5055d70a5de8324f4c321f2436d042b65c06
Python 3.12.6, Qt 6.7.3, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
```

The crash appeared in [`QDialog::reject()`](https://github.com/FreeCAD/FreeCAD/blob/146f5055d70a5de8324f4c321f2436d042b65c06/src/Gui/DlgAddPropertyVarSet.cpp#L526). I noticed strange behavior with the transactions as reported in #17486, so I assumed that was playing a role. While trying to acquire debug symbols for the qt libraries, I need to upgrade the machine which upgraded Qt to 6.8. After that I couldn't reproduce anymore. The configuration for that machine is (it is on the `main` branch instead of the release branch):

```
OS: Arch Linux (xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39062 (Git)
Build type: Release
Branch: main
Hash: c5c64a9c70f21c4c4c2ad294645993f9a60342d8
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
```

To figure out what was going on, I used a different computer to get back to this configuration:

```
OS: Arch Linux (xcb)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38997 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: 146f5055d70a5de8324f4c321f2436d042b65c06
Python 3.12.6, Qt 6.7.3, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
```

Unfortunately, on this computer I can't reproduce anymore. I'm now assuming that it is a Qt bug that was somehow triggered. In any case, https://github.com/FreeCAD/FreeCAD/pull/17537 improves on #17486 and with that PR I didn't notice the behavior that this issue describes when my computer was in a state in which I could reproduce it.

If anyone can reproduce and provide a stacktrace, preferably into the Qt libraries as well, that would be very helpful.

wwmayer, 2024-10-28:
> The crash appeared in QDialog::reject()

Yes, but the dialog has already been destroyed before calling this method. It's because the owner of the dialog is ViewProviderVarSet and by calling abortTransaction() inside DlgAddPropertyVarSet::reject() the VarSet object and the view provider will be destroyed and with it the dialog.

So, calling QDialog::reject() afterwards causes undefined behaviour -- either by accessing a member variable or by trying to double destruct the dialog.

> I noticed strange behavior with the transactions as reported in https://github.com/FreeCAD/FreeCAD/issues/17486, so I assumed that was playing a role.

Not really.

> I'm now assuming that it is a Qt bug

Definitely not!

A way (not nice but it works) to check for this is to write:
void DlgAddPropertyVarSet::reject()
{
...
    QPointer<QObject> self(this);
    // a transaction is not pending if a name has not been determined.
    if (doc->hasPendingTransaction()) {
        doc->abortTransaction();
    }
    if (self) {
        QDialog::reject();
    }
}

wwmayer, 2024-10-28:
The PR was merged while writing the comment. After testing it I can confirm that the crash is fixed.

closed 2024-10-28 by wwmayer

pieterhijma, 2024-10-28:
> The PR was merged while writing the comment. After testing it I can confirm that the crash is fixed.

Really? The PR was written without me realizing all that you wrote in the comment above, so if it really fixes this issue, then it feels as if I was lucky. In any case, as we speak, I'm looking into it more closely and if needed, I will reopen and create a new PR.

wwmayer, 2024-10-28:
Adding commitCommand to StdCmdVarSet makes sure that the creation of the VarSet object isn't part of a pending transaction and thus aborting it will no longer destroy the VarSet object.

pieterhijma, 2024-10-28:
Right, the other mitigation is that it only tracks the "Add property to VarSet" transactions and therefore will never abort the "create VarSet" transaction.

Thanks for giving me all those insights.

PR #17537 (2024-10-28): Gui: Make VarSet dialog modal
Also tracks its own transactions to prevent interfering with other transactions.

This attempts to fix #17475, but I can't reproduce that anymore, so I'm not sure about that. When I was still on the configuration that triggered the bug, this PR did not show the same behavior.

This fixes #17486.

I see two viable solutions for #17486. One is using a transaction lock, but with a non-modal dialog, this can block other transactions that may lead to unexpected behavior for the user. The other one is making the dialog modal, which is what I chose in the end. The dialog [... cut]