FreeCAD/FreeCAD #17487: CI: C++ tests falure on macOS -- check fails during cleanup phase

AIRCAP, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In https://github.com/FreeCAD/FreeCAD/actions/runs/11517148162/job/32061268593?pr=17469

`[----------] Global test environment tear-down
[==========] 15 tests from 2 test suites ran. (101 ms total)
[  PASSED  ] 15 tests.
Run result=$(sed -ne "/Global test environment tear-down/,/^$/{/^$/d;p}" /tmp/report/points_gtest_test_log.txt)
sed: couldn't flush stdout: Broken pipe
Error: The action has timed out.
Run # Print global result
Error: The action has timed out.`

### Full version info

```shell
N/A
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-25:
affects #17469 right now, possibly others

AIRCAP, 2024-10-25:
#17479 affected too

AIRCAP, 2024-10-25:
#17450 too

luzpaz, 2024-10-25:
cc @adrianinsaval