FreeCAD/FreeCAD #17491: Viewport completely white/black

guiweber, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm seeing all viewports completely white (they appear black in screenshots) on v. 1.0.0RC2. Only the start page when opening the application looks fine.

I've cross checked with v. 21.2 on the same PC and I am not seeing the issue. All viewports are rendered properly with hardware acceleration and VBO active.

On 1.0.0RC2 I've tried to fix the issue through the following steps (without success):
- Use software OpenGL
- Uncheck use OpenGL VBO
- Disable Anti-Aliasing
- Change the theme to dark, white or classic

Let me know if I can provide more information to help pinpoint the issue.

### Full version info

```shell
OS: CachyOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-25:
Could you try with the latest weekly https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
And maybe also restart FreeCAD in safe mode (new weekly: Help>Restart in safe mode)

guiweber, 2024-10-26:
So I've done a bit more testing and it seems that any version run through an AppImage doesn't cause the issue (tested 21.2, 1.0.0RC2 and  1.1.0 (build 39024).

So the issue has something to do with the version published on the Arch Linux Extra repository. I tried to uninstall and reinstall, making sure that there were no errors, but the issue is still present.

luzpaz, 2024-10-26:
@guiweber good to know, that helps us to narrow things down. There are other Arch/Manjaro users who run that Arch build here. Can you send a message to the Arch Linux Extra repository build comment section and let them know you opened a bug here so they could help troubleshoot it ?

benj5378, 2024-10-27:
@guiweber can you please try with software opengl (choose in preferences). And also check the opengl version

guiweber, 2024-10-27:
@luzpaz  Absolutely, but I'm not sure what you're referring to. Do you mean the [FreeCAD-Bundle repo](https://github.com/FreeCAD/FreeCAD-Bundle/issues) the [arch linux package gitlab](https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues)?

@benj5378 I have tried software opengl but the issue persists. I however get stdout errors when running through the terminal. They are always present and the same no matter if software opengl is selected or not.

When opening the preference panel:

> (qt.qpa.wayland) eglSwapBuffers failed with 0x300d, surface: 0x0

When opening a viewport:

> Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
> Coin warning in SoGLPolygonOffsetElement::updategl(): OpenGL driver doesn't support z-buffer offsetting

guiweber, 2024-10-27:
@benj5378 OpenGL version string: 4.6.0 NVIDIA 560.35.03

benj5378, 2024-10-27:
@guiweber. this error seems to explain the issue,
```
Coin warning in cc_glglue_instance(): Error when setting up the GL context. This can happen if there is no current context, or if the context has been set up incorrectly.
```

Please try `QT_QPA_PLATFORM=xcb ./bin/freecad` (or `QT_QPA_PLATFORM=xcb freecad`) in terminal.

If this doesn't work, could you please supply the full output of `glxinfo | grep version`? And change https://github.com/FreeCAD/FreeCAD/blob/c5c64a9c70f21c4c4c2ad294645993f9a60342d8/src/Gui/View3DInventor.cpp#L104-L111 to
```cpp
    f.setVersion(3, 4);  // Perhaps also try with f.setVersion(3, 3);
    if (samples > 1) {
        glformat = true;
        f.setSamples(samples);
    }
    else if (samples > 0) {
        smoothing = true;
    }
    glformat = true;
```
and then run try running with `./bin/freecad` and `QT_QPA_PLATFORM=xcb ./bin/freecad`.

If this doesn't work, please compile https://github.com/coin3d/coin/pull/531 and link it to FreeCAD by setting https://github.com/FreeCAD/FreeCAD/blob/c5c64a9c70f21c4c4c2ad294645993f9a60342d8/cMake/FreeCAD_Helpers/SetupCoin3D.cmake#L9-L18 to
```
    # Try MODULE mode
    # comment out this line! find_package(Coin3D)
    if (NOT COIN3D_FOUND)
        # Try CONFIG mode
        find_package(Coin CONFIG REQUIRED)
        if (Coin_FOUND)
            set(COIN3D_INCLUDE_DIRS "/home/bensay/Development/coin-build/coin_egl_build/include/")
            set(COIN3D_LIBRARIES "/home/bensay/Development/coin-build/coin_egl_build/lib/libCoin.so")
        endif ()
    ENDIF ()
```

Keep me update what kind of different errors you receive in terminal, through all these different steps. All errors containing `coin`, `context`, `egl`, `opengl`, `glue`, `gl`, `glx`.

luzpaz, 2024-10-28:
> Do you mean the [FreeCAD-Bundle repo](https://github.com/FreeCAD/FreeCAD-Bundle/issues) the [arch linux package gitlab](https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues)?

@guiweber [arch linux package gitlab](https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues) please. TIA!

guiweber, 2024-10-28:
@benj5378 Running the app with `QT_QPA_PLATFORM=xcb freecad` fixes the issue!

luzpaz, 2024-10-28:
> Running the app with QT_QPA_PLATFORM=xcb freecad fixes the issue!

Does that make this a Wayland related issue ?

benj5378, 2024-10-28:
@luzpaz, yes, wayland related due to coin not supporting EGL. We can close this issue - just a duplicate of others

benj5378, 2024-10-28:
We might want to tell Arch Repo people to make this flag default for all users.

adrianinsaval, 2024-11-05:
sounds to me like in freecad we should be forcing use of xcb platform rather than letting users choose

guiweber, 2025-08-25:
Still present...

Lucretia, 2026-10-22:
I've just had this, except it was the entire window, not the window decorations.