FreeCAD/FreeCAD #17497: Part: OCCT cannot cut from intersecting parts (for example in compounds)

AIRCAP, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

When you try to (boolean) cut a shape from a compound of shapes that are intersecting each other, it fails.

It typically (but not always) only fails in the areas where the parts are actually intersecting.  While the correct surfaces are created, there are also additional wrong surfaces - the resulting part is self intersecting and fails all geometry checks.

This also affects boolean fragments if the fragments are grouped in different containers but have intersecting shapes in the same sub-container.

I don't know if this is a bug in OCCT or just a limitation of OCCT. Either way it is clearly an OCCT issue.

This report is just so the issue is documented.

### Full version info

```shell
OCC 7.6.3
```

### Subproject(s) affected?

Core

### Anything else?

Possible workaround:

FreeCAD could unpack compounds used in boolean cut's - cut each shape individually, and then re-assemble the cut parts into compounds to avoid this issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-25:
> OCC 7.6.3  

Just wondering:  Latest OCCT is 7.8.1  
Any reason you haven't tested it on that?

AIRCAP, 2024-10-25:
> > OCC 7.6.3
> 
> Just wondering: Latest OCCT is 7.8.1 Any reason you haven't tested it on that?

will have to compile from src to test that.
I just testes latest weekly with 7.7.1 - same issue

AIRCAP, 2024-10-25:
OCCT 7.3.0 same issue

AIRCAP, 2024-10-25:
@luzpaz Just tested the latest - issue still the same:

```
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39036 +5 (Git)
Build type: Debug
Branch: regression
Hash: 26729d47e7efca0c7da5388f446754f5e12e6774
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.2.dev
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.12
  * CfdOF 1.27.10
  * Curves 0.6.48
  * AirPlaneDesign 0.4.1
```

AIRCAP, 2024-10-25:
Should we do that? as in unpack compounds for cutting?

With the FCBRepAlgoAPI wrappers you could just put it in the wrapper and have it seamless

FlachyJoe, 2024-10-26:
> I don't know if this is a bug in OCCT or just a limitation of OCCT. Either way it is clearly an OCCT issue.

Should be asked at https://dev.opencascade.org/forums

> Possible workaround:
> 
> FreeCAD could unpack compounds used in boolean cut's - cut each shape individually, and then re-assemble the cut parts into compounds to avoid this issue.

It's sounds good.

luzpaz, 2024-10-26:
cc @dpasukhi if you have a spare moment could you weigh-in ?

dpasukhi, 2024-10-26:
I will ask modeling team about the limitations itself or it is a bug and possible workaround.
Originally, it must be a fuse operating before. Because geometry itself is not easy in that case. Probably even we will not be able to fully fix that problem for all intersection cases. I would recommend to fuse in case of intersection.

I'm afraid that at the moment I have no available developers to fix or deep analyze the issue at least for closest month :(
Additionally I will create an issue on our side (in GitHub). Or you are free to create from your side.

luzpaz, 2024-10-27:
@AIRCAP should we create the issue or should we ask dpasukhi to open a ticket on OCCT ?

AIRCAP, 2024-10-27:
I think opening a ticket might make sense just to document the issue. I do not have any account on OCCT so it would be better if dpasukhi did it. 

It's in the end up to the OCCT devs if they want to treat this issue as a limitation of OCCT or as a bug that needs to be fixed.

regardless it doesn't have to be a limitation of FreeCAD, since the workaround is relatively easy.

its easy, we only use a relatively small subset of boolean CUT's interface.

CUT in OCCT can CUT multiple shapes with multiple tools, but FreeCAD only ever calls CUT with a single argument shape and a single tool (shich can however be compounds)
CUT in OCCT also has CUT21 (which is a reverse cut - cutting the tools with the arguments if I understood that right) FreeCAD doesn't use that either.

Therefore for the subset that freecad uses, fixing this is (probably) easier than for OCCT in general.

It should be noted that we can only fix compounds that way, not broken COMPSOLID or other self intersecting shapes.

but trying to cut a hole into compound of overlapping shapes is a useful feature for example for Cfd - where the compounds represent the volume of fluid to be computed at different refinement levels - and the shape cut from them is the actual design to be analyzed.

siilarly FEM can also have cases where compound solids are intersecting

I should note there is a related issue (or is it the same) that fails even worse. cutting a shape with a compound of intersecting tools:

[image]

cutting the 3 cylinders with 2 overlapping cubes completely breaks the middle one - one end is completely missing, one end has its normals flipped, and the middle part is inversed (the part cut a single time is still standing, the part cut out twice is missing)

that too could be fixed, by repeated cuts for each shape in a compound (again of course recursively, depending how deep the compound madness goes)

I have the fix for the first case implemented:

[image]

the result is now a compound consisting of

[image]
and
[image]

working on the second case (intersecting tools)

AIRCAP, 2024-10-27:
heres the code to fix the compound base

```
diff --git a/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp b/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp
index 96af048107..7c888187b2 100644
--- a/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp
+++ b/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp
@@ -30,7 +30,9 @@
 #include <BRepBndLib.hxx>
 #include <Bnd_Box.hxx>
 #include <BRepCheck_Analyzer.hxx>
+#include <BRep_Builder.hxx>
 #include <TopoDS_Shape.hxx>
+#include <TopoDS_Iterator.hxx>
 #include <Precision.hxx>
 #include <FuzzyHelper.h>
 
@@ -78,3 +80,33 @@ void FCBRepAlgoAPIHelper::setAutoFuzzy(BRepAlgoAPI_BuilderAlgo* op) {
         BRepBndLib::Add(it.Value(), bounds);
     op->SetFuzzyValue(Part::FuzzyHelper::getBooleanFuzzy() * sqrt(bounds.SquareExtent()) * Precision::Confusion());
 }
+
+void FCBRepAlgoAPI_BooleanOperation::Build(const Message_ProgressRange& theRange) {
+
+    if (myOperation==BOPAlgo_CUT && myArguments.Size()==1 && myArguments.First().ShapeType() == TopAbs_COMPOUND) {
+        myOriginalArguments = myArguments;
+        myShape = RecursiveCutCompound(myOriginalArguments.First(), theRange);
+    } else {
+        return BRepAlgoAPI_BooleanOperation::Build(theRange);
+    }
+
+}
+
+const TopoDS_Shape FCBRepAlgoAPI_BooleanOperation::RecursiveCutCompound(const TopoDS_Shape& theArgument, const Message_ProgressRange& theRange) {
+    BRep_Builder builder;
+    TopoDS_Compound comp;
+    builder.MakeCompound(comp);
+    TopoDS_Iterator it(theArgument);
+    for (; it.More(); it.Next()) {
+        TopTools_ListOfShape currentArguments;
+        currentArguments.Append(it.Value());
+        myArguments = currentArguments;
+        Build(theRange);
+        if (IsDone()) {
+            builder.Add(comp, myShape);
+        } else {
+            return TopoDS_Shape();
+        }
+    }
+    return comp;
+}
diff --git a/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.h b/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.h
index dcd3451e14..a0e392c00e 100644
--- a/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.h
+++ b/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.h
@@ -49,6 +49,8 @@ public:
     // set fuzzyness based on size
     void setAutoFuzzy();
 
+    Standard_EXPORT virtual void Build(const Message_ProgressRange& theRange = Message_ProgressRange()) Standard_OVERRIDE;
+
 protected: //! @name Constructors
 
   //! Constructor to perform Boolean operation on only two arguments.
@@ -57,5 +59,10 @@ protected: //! @name Constructors
                                                const TopoDS_Shape& theS2,
                                                const BOPAlgo_Operation theOperation);
 
+
+private:
+  TopTools_ListOfShape myOriginalArguments;
+  Standard_EXPORT const TopoDS_Shape RecursiveCutCompound(const TopoDS_Shape& theArgument, const Message_ProgressRange& theRange);
+
 };
 #endif

```

commit 648bb8f5 (2025-03-17): fix #17497
closed 2025-02-14 by chennes