FreeCAD/FreeCAD #17500: FreeCAD Ribbon dissapears when autohide is enabled or collapsed (Only FreeCAD 1.0 QT5 & QT6)

APEbbers, 2024-10-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When autohide is enabled or the collapse button is clicked a second time. The Ribbon disappears.
This issue is only with FreeCAD 1.0 with either QT5 or QT6

Link to issue in my repository: [https://github.com/APEbbers/FreeCAD-Ribbon/issues/54]

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Netherlands (en_NL)
Stylesheet/Theme/QtStyle: ExtremeProDark.qss/FreeCAD Dark/
Installed mods: 
  * A2plus 0.4.68 (Disabled)
  * Assembly3 0.12.2 (Disabled)
  * Assembly4 0.50.15
  * BillOfMaterials 1.0.1
  * boltsfc 2022.11.5
  * CfdOF 1.27.12
  * DynamicData 2.68.0
  * ExtremeProDark 2.9.0
  * fasteners 0.5.29
  * fasteners.backup1729411277.8118737 0.5.29 (Disabled)
  * FEMbyGEN 2.2.0 (Disabled)
  * FEM_FrontISTR 0.1.0
  * frame 0.1.1
  * FreeCAD-Beginner-Assistant 1.0.0
  * FreeCAD-Ribbon 1.3.5
  * freecad.gears 1.2.0
  * IconThemes (Disabled)
  * Launcher
  * NikraDAP 2.0.0
  * OpenTheme 2024.9.1
  * PieMenu 1.8.1 (Disabled)
  * Plot 2024.10.23
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.9.25
  * sheetmetal 0.5.3
  * ThreadProfile 1.92.0
  * WorkFeature-WB
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

https://github.com/APEbbers/FreeCAD-Ribbon

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-26:
@APEbbers could we trouble you for a screencast of the issue in action ?

APEbbers, 2024-10-26:
Of course. Here is a capture of the problem:

https://github.com/user-attachments/assets/fd6076a8-28b7-466f-a9a2-1019169de794

I suspect it has to do with the "Dock window overlay" settings. Currently the Ribbon is loaded as a QDockWidget.

I'm currently looking into replacing the MenuBar with the ribbon instead.

Code as it is now:
[image]

Code which i'm looking into:
[image]

luzpaz, 2024-10-26:
@kadet1090 any chance you could glimpse at this overlay related issue ?

APEbbers, 2024-10-28:
Hi,

I went futher with the investigation and my theory is that both the overlay and my Ribbon had "EnterEvent"/"LeaveEvent" functions. 

In my Develop branch I have a version ready which isn't loaded as a QDockWidget. This way, the ribbon doesn't need the functions mentioned. Also it is not possible to make the ribbon transparant by (accidently) toggle the overlay, because it isn't a dockwidget.

@luzpaz or @kadet1090: can you check if this solves the issue? the link to my branch is:
[https://github.com/APEbbers/FreeCAD-Ribbon.git](url)

thanks in advance!

kind regards

luzpaz, 2024-10-29:
This links to the FreeCAD-Ribbon `main` branch. Is that your intention OR did you intend to link to `54-ribbon-dissapears-when-autohide-is-enabled-or-collapsed-only-freecad-10-qt5-qt6` ?

APEbbers, 2024-10-29:
sorry, no i actually intended to link to the develop branch. [https://github.com/APEbbers/FreeCAD-Ribbon/tree/Develop](url)

hyarion, 2024-10-30:
> I'm currently looking into replacing the MenuBar with the ribbon instead.

Don't forget that macOS (and some *nix window managers) has a dedicated space for the menu bar outside the window (top of screen on mac). Removing the menu bar on those system will probably just remove the freecad items, not free up any space.

hyarion, 2024-10-30:
@APEbbers In the comments above it sounds like the reported issue could be a bug in your code.
Is there a freecad bug here? If so, can you create a minimal test case which shows the issue?

APEbbers, 2024-10-30:
Hi @hyarion ,

Well, i don't have an issue in FreeCAD 0.21. But I believe the issue is related with toggling the overlays in FreeCAD.
In my develop branch, I have a solution that solves that problem by loading the Ribbon as a QMenubar in the mainwindow of FreeCAD.

I don't have access to a mac to test this in macOS. Do you have installed a FreeCAD 1.0dev version perhaps? If so are you willing to test the Develop version ([[https://github.com/APEbbers/FreeCAD-Ribbon/tree/Develop](https://github.com/FreeCAD/FreeCAD/issues/url)](url) to see if this solution works or not?

kind regards,

hyarion, 2024-10-30:
> I don't have access to a mac to test this in macOS. Do you have installed a FreeCAD 1.0dev version perhaps? If so are you willing to test the Develop version

`19:03:13  During initialization the error "No module named 'Quartz'" occurred in /Users/nauck/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./InitGui.py`

APEbbers, 2024-10-30:
Hi,

Quartz is used by the keyboard package. I have removed it from the develop branch. Can you give it a try again?
it is still on the main branch, so perhabs it is easier to download a zipfile from my develop branch instead.

hyarion, 2024-10-30:
Sorry, I cloned the wrong branch.
On develop, I don't see any tool bar at all on macos, no ribbon, no nothing.
[image]

APEbbers, 2024-10-30:
hmm, that's not good. perhaps, the ribbon must be loaded differently per OS. I'm currently downloading an ISO image with macOS to setup an test environment.

Hopefully I can then see what happening. Thanks for testing anyway!

APEbbers, 2024-11-02:
So It appears that there where signals interfering with each other. I have solved the issue by adding these lines of code to my "__init__" function:
[image]

This solves the main issue. You can still toggle "overlay", which is not Ideal for the ribbon but that is not as urgent as the ribbon disappearing entirely when hovering over it. Besides, you can toggle it back.

I will still setup a macOS enviroment to see if i can load the ribbon differently, but setting this proves a bit of a pain.
For now I'm closing this issue. Thanks for all the help.

kind regards,

APEbbers, 2024-11-03:
Hi all,

I've found the exact problem and a solution. Can anyone test this solution with my develop branch? ([[https://github.com/APEbbers/FreeCAD-Ribbon/tree/Develop](https://github.com/FreeCAD/FreeCAD/issues/url)](url)

The root of the problem is specific related to the EnterEvent function. The overlay function has also a EnterEvent which is also triggered and caused this the strange behaviour. 

The solution I have now is that I have disabled the EnterEvent function for the Ribbon. (Only for FC 1.0). This means that you have to click on a tab to show the ribbon again. Personally, I don't think that is a problem. It might be even seen as an improvement because it avoids accidently showing the ribben. (results in more predictable behaviour)
With this solution, the Ribbon remains a QDockWidget, which should work on macOS. (I'm still setting up a test environment for macOS, but that proves to be more difficult than I thought)

Thanks in advance!

APEbbers, 2024-12-16:
Hi, I've solved the issue by disabling the native overlay function via parameters. I created a custom function that enables overlay for all panels except the ribbon.

> Sorry, I cloned the wrong branch. On develop, I don't see any tool bar at all on macos, no ribbon, no nothing. 
From version 1.4.16 onwards, macOS is now supported.

kind regards,

closed 2024-12-16 by APEbbers