FreeCAD/FreeCAD #17503: Loft command

DrAvriLev, 2024-10-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to create a loft I got the following error:
BRep_API:command not done

### Full version info

```shell
OS: Windows 11 build 26120
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.50
  * lattice2 1.0.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-26:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- Test if the issue is still present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

luzpaz, 2024-10-26:
Perhaps related to #16630 ?

FEA-eng, 2024-10-26:
A more meaningful title like "Loft - BRep_API:command not done" would be nice too.

AIRCAP, 2024-10-28:
> Perhaps related to #16630 ?

yeah it might be a duplicate. if it is, its fixed in dev

maxwxyz, 2024-11-01:
@DrAvriLev please test in the latest weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds 
If there is still an issue, comment and I'll reopen the issue.

closed 2024-11-01 by maxwxyz

DrAvriLev, 2024-11-01:
Sorry,
how do I test that build?

Kind regards,

Dr/ Avri Lev
cell =972-53-456.5368

________________________________
מאת: Max Wilfinger ***@***.***>
‏‏נשלח: יום שישי 01 נובמבר 2024 12:03
‏‏אל: FreeCAD/FreeCAD ***@***.***>
עותק: DrAvriLev ***@***.***>; Mention ***@***.***>
‏‏נושא: Re: [FreeCAD/FreeCAD] Loft returns `BRep_API:command not done` error (Issue #17503)

@DrAvriLev<https://github.com/DrAvriLev> please test in the latest weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
If there is still an issue, comment and I'll reopen the issue.

maxwxyz, 2024-11-01:
Download the one for Windows, unzip and start the FreeCAD.exe in the bin folder.

DrAvriLev, 2024-11-01:
Thank you so much

________________________________
מאת: Max Wilfinger ***@***.***>
‏‏נשלח: יום שישי 01 נובמבר 2024 16:02
‏‏אל: FreeCAD/FreeCAD ***@***.***>
עותק: DrAvriLev ***@***.***>; Mention ***@***.***>
‏‏נושא: Re: [FreeCAD/FreeCAD] Loft returns `BRep_API:command not done` error (Issue #17503)

Download the one for Windows, unzip and start the FreeCAD.exe in the bin folder.

—

T

philjynx, 2024-11-08:
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fcgear
  * FreeCAD-themes 2024.7.24
  * Help 1.0.3
  * HiResMonitor
  * kicadStepUpMod 11.4.1
  * Manipulator 1.5.7
  * sheetmetal 0.5.3

I have the same error message in V1RC2 (which I am now using all the time) 

**the error does not happen in the above weekly build.**